balena-io/etcher #2849: UI back button inconsistent cursor [Mac OS]

peterblazejewicz, 2019-07-07:
See the behaviour on the screen record:

[image]

- **Etcher version:** 
1.5.51
- **Operating system and architecture:** 
Mac OS Mojave, 10.14.5 (18F132)
- **Image flashed:** 
n/a
- **Do you see any meaningful error information in the DevTools?** 
Yes, the detail is implemented using native button, but trying to force it acting like a hyperlink:

[image]

amrsekilly, 2019-08-05:
Can I open a PR with a fix for this?

thundron, 2019-08-05:
@amrsekilly Indeed! PRs are more than welcome for anyone that wants to contribute

amrsekilly, 2019-08-05:
Thanks, I'll submit a PR with a fix shortly!