balena-io/etcher #2850: Provide downloaded software signature verification

Jack15911, 2019-07-09:
balenaEtcher 1.5.51 and all subsequent
macOS, Windows, Linux

All or most of the other open source software provides public key verification of software and checksums values to ensure the validity of the downloaded file. Issue #2826 addresses the latter checksums issue. That leaves the digital signature issue remaining. Both are necessary.

The public/signing key verifies that the downloaded software is the actual software the Team produced. There are several ways to approach it; any will serve as long as (1) each package download is covered, and (2) balenaEtcher provides instructions.

MacOS and (I believe) Windows require software packages to be digitally signed before they may be installed.  I don't believe Linux does. However, it might be possible for a person is phished to another site in which the binary is signed, but not by balena-Etcher.  Another reason for checking digital signatures would be to avoid downloading and installing a good balenaEtcher package that has been encapsulated with malware and digitally signed; it would pass the checksums, install and run, and still inject malware onto a system. 

I have personal experience with other open source software - KeePass variants, OpenVPN, nmap, gpgTools, Ubuntu, Mint among them, and all of them provide PK signature verification, as well as artifact checksums validation. 

Easy verification of signatures and validation of checksums cover separate issues, and both are necessary.

rajuashok, 2020-12-31:
Is there any update here? I too am looking for a signature file to verify binary.

jtheil, 2021-02-03:
Is there any update on this? It is uncomfortable not being able to have the keys.

thundron, 2021-02-03:
We already sign the packages for the obvious reasons you explained, so other than that what is it that you're suggesting here exactly?

lurch, 2021-02-03:
@thundron It's not something I'm bothered about myself (I'm always careful about which websites I download things from), but _perhaps_ people are looking for something similar to what VirtualBox does? :shrug: 
https://www.virtualbox.org/wiki/Linux_Downloads

rajuashok, 2021-02-03:
Something similar to VirtualBox would be great. Or even better Trezor Suite: https://suite.trezor.io/. Notice the signing key and signature files right underneath the download button.

Jack15911, 2021-02-04:
> @thundron It's not something I'm bothered about myself (I'm always careful about which websites I download things from), but _perhaps_ people are looking for something similar to what VirtualBox does? 🤷
> https://www.virtualbox.org/wiki/Linux_Downloads

When we download a package we want to authenticate not just that the package has been signed, but that it has been signed by the proper person at balena-etcher; that's done through the gpg process of adding the proper signature to our files then checking it. Otherwise we might have a package signed by someone we don't trust.

We also want to verify that the package has not been altered during the download, either by accident or intent. We do this with the shasum 256 checksums process. It doesn't require gpg, but sometimes it's easier.

I think the problem is that balena-etcher doesn't allow a us to authenticate the signature, and doesn't provide a manual package checksums verification. If we have to run it though a "black box" process that (I, at least) don't fully understand, then we don't have confidence in the result.

Jack15911, 2021-02-04:
> We already sign the packages for the obvious reasons you explained, so other than that what is it that you're suggesting here exactly?
> 
> EDIT: we also have checksums available for every major package type in each release, should that be of concern to anyone

When we download a package we want to authenticate not just that the package has been signed, but that it has been signed by the proper person at balena-etcher; that's done through the gpg process of adding the proper signature to our files then checking it. Otherwise we might have a package signed by someone we don't trust.

We also want to verify that the package has not been altered during the download, either by accident or intent. We do this with the shasum 256 checksums process. It doesn't require gpg, but sometimes it's easier.

I think the problem is that balena-etcher doesn't allow a us to authenticate the signature, and doesn't provide a manual package checksums verification. If we have to run it though a "black box" process that (I, at least) don't fully understand, then we don't have confidence in the result.

thundron, 2021-05-17:
please see https://github.com/balena-io/etcher/issues/2826 , we'll track the issue there

closed 2021-05-17 by thundron

sig-noi, 2023-09-06:
Just went through this process. Still more work than it should be since balena doesn’t display shasums in a typical location or format (256, hex). Also not secure since there is no gpg signature to authenticate the origin of the shasum file. In any case, I used the mac-lates.yml to find the base64 signature, used this utility to get hex https://base64.guru/converter/decode/hex , then compared. I also used the `codesign -dv --verbose=4 /Applications/balenaEtcher.app` to view the app’s signature. The shasums match, and the file is signed by Balena Ltd (66H43P8FRG) however I find few references to this online to verify that it is the correct ID. It would be ideal to have Balena’s signing ID prominently displayed for verification of the file’s signature, as well as the shasum in the more common hex format. Additionally agree with above commenters that the more common gpg-signed approach would be much simpler and preferable.