balena-io/etcher #2851: Benchmarking Etcher

lurch, 2019-07-09:
- **Etcher version:** 1.5.51 AppImage
- **Operating system and architecture:** Ubuntu 16.04 x64
- **Image flashed:** 2019-06-20-raspbian-buster-full
- **Do you see any meaningful error information in the DevTools?** n/a

I recently bought myself a [250GB Samsung T5 USB3 SSD](https://www.samsung.com/semiconductor/minisite/ssd/product/portable/t5/) which is *very* fast, so before I start using it for other purposes I thought I'd do a quick bit of benchmarking in Etcher :wink: :nerd_face: 

My desktop PC has a Core i7-5820K CPU and 32GB of RAM, running the OS from an SSD, and I'm using the motherboard's USB3 ports, so there shouldn't be any OS/hardware limiting factors.

If I try to write a compressed image i.e. `2019-06-20-raspbian-buster-full.zip` to the drive, Etcher writes to the drive at a pretty decent 60 MB/s, and watching the CPU monitor in the background, none of the CPUs go above 40% usage.

If I uncompress the image first i.e. to `2019-06-20-raspbian-buster-full.img` then Etcher writes to the drive at a much more impressive 150 MB/s! I was surprised at just how _much_ quicker it was with the image uncompressed first, given that none of the CPUs appear to be the throttling factor.

However even that speed seems "slow" when compared to the dd command - `sudo dd if=~/Downloads/2019-06-20-raspbian-buster-full.img bs=4M of=/dev/sdc conv=fdatasync` writes to the drive at 340 MB/s. I wasn't expecting the GUI of Etcher to add so much overhead compared to the command-line of dd :confused: 

I repeated each of the above tests 3 times to ensure I was seeing reliable figures and not statistical flukes.

Strictly speaking this isn't a "problem" so feel free to close this issue, but I thought you might find the figures interesting. Especially given the "Insane Speeds" you're aiming for with https://www.balena.io/etcher/pro  :racing_car:

brianread108, 2019-08-09:
I am running Fedora 30 on a 16gb AMD Ryzen system and since upgrading etcher I am see very slow access to the SSD. I am getting less that 1Mb/s writing or validating.  The Previous version (on Fedora 29) wrote a great deal faster (I never worried about the speed!).

lurch, 2019-08-09:
@brianread108 Maybe it's worth timing some `dd` writes to the SSD as well, to check that it's new Etcher to blame and not new Fedora?

brianread108, 2019-08-09:
mm, just run a dd which took 1.5 hours - under 1Mb/s. I need to think about fedora 30 rather thna etcher I guess.