balena-io/etcher #2854: Something Went Wrong - with every ISO I try to burn

Cylindric, 2019-07-14:
- 1.5.51
- Windows 10
- hassos_rpi3-2.12.img

I do get an error in the console, not sure if it's useful or not:

```
{
  "name": "Error",
  "message": "Command failed: C:\\Users\\mark\\AppData\\Local\\Programs\balena-etcher\balenaEtcher.exe C:\\Users\\mark\\AppData\\Local\\Programs\balena-etcher\resources\\app.asar\nELECTRON_ASAR.js:112\r\n      throw error\r\n      ^\r\n\r\nError: ENOENT, node_modules\\xxhash\build\\Release\\hash.node not found in C:\\Users\\mark\\AppData\\Local\\Programs\balena-etcher\resources\\app.asar\r\n    at notFoundError (ELECTRON_ASAR.js:108:19)\r\n    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:176:9)\r\n    at Module.load (internal/modules/cjs/loader.js:620:32)\r\n    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)\r\n    at Function.Module._load (internal/modules/cjs/loader.js:551:3)\r\n    at Module.require (internal/modules/cjs/loader.js:658:17)\r\n    at require (internal/modules/cjs/helpers.js:20:18)\r\n    at Object.<anonymous> (C:\\Users\\mark\\AppData\\Local\\Programs\balena-etcher\resources\\app.asar\node_modules\\xxhash\\lib\\xxhash.js:4:13)\r\n    at Object.<anonymous> (C:\\Users\\mark\\AppData\\Local\\Programs\balena-etcher\resources\\app.asar\node_modules\\xxhash\\lib\\xxhash.js:52:3)\r\n    at Module._compile (internal/modules/cjs/loader.js:711:30)\r\n    at Object.Module._extensions..js (internal/modules/cjs/loader.js:722:10)\r\n    at Module.load (internal/modules/cjs/loader.js:620:32)\r\n    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)\r\n    at Function.Module._load (internal/modules/cjs/loader.js:551:3)\r\n    at Module.require (internal/modules/cjs/loader.js:658:17)\r\n    at require (internal/modules/cjs/helpers.js:20:18)\r\n",
  "cause": {
    "killed": false,
    "code": 1,
    "signal": null,
    "cmd": "app.asar"
  },
  "isOperational": true,
  "stack": "Error: Command failed: C:\\Users\\mark\\AppData\\Local\\Programs\balena-etcher\balenaEtcher.exe C:\\Users\\mark\\AppData\\Local\\Programs\balena-etcher\resources\\app.asar\nELECTRON_ASAR.js:112\r\n      throw error\r\n      ^\r\n\r\nError: ENOENT, node_modules\\xxhash\build\\Release\\hash.node not found in C:\\Users\\mark\\AppData\\Local\\Programs\balena-etcher\resources\\app.asar\r\n    at notFoundError (ELECTRON_ASAR.js:108:19)\r\n    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:176:9)\r\n    at Module.load (internal/modules/cjs/loader.js:620:32)\r\n    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)\r\n    at Function.Module._load (internal/modules/cjs/loader.js:551:3)\r\n    at Module.require (internal/modules/cjs/loader.js:658:17)\r\n    at require (internal/modules/cjs/helpers.js:20:18)\r\n    at Object.<anonymous> (C:\\Users\\mark\\AppData\\Local\\Programs\balena-etcher\resources\\app.asar\node_modules\\xxhash\\lib\\xxhash.js:4:13)\r\n    at Object.<anonymous> (C:\\Users\\mark\\AppData\\Local\\Programs\balena-etcher\resources\\app.asar\node_modules\\xxhash\\lib\\xxhash.js:52:3)\r\n    at Module._compile (internal/modules/cjs/loader.js:711:30)\r\n    at Object.Module._extensions..js (internal/modules/cjs/loader.js:722:10)\r\n    at Module.load (internal/modules/cjs/loader.js:620:32)\r\n    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)\r\n    at Function.Module._load (internal/modules/cjs/loader.js:551:3)\r\n    at Module.require (internal/modules/cjs/loader.js:658:17)\r\n    at require (internal/modules/cjs/helpers.js:20:18)\r\n\n    at ChildProcess.exithandler (child_process.js:303:12)\n    at ChildProcess.emit (events.js:182:13)\n    at maybeClose (internal/child_process.js:961:16)\n    at Process.ChildProcess._handle.onexit (internal/child_process.js:248:5)",
  "killed": false,
  "code": 1,
  "signal": null,
  "cmd": "app.asar",
  "image": "hassos_rpi3-2.12.img"
}
```

This is happening 100% of the time, with any image I try.

DenjW, 2019-07-16:
I have just installed Appimage 1.5.51 64bit to Linux Lite 4.4 and have experienced exactly the same problem: After completion of Flash I get a message after validation (or during - not sure) that "Something went wrong" in red box.  First time yesterday I used the same iso image and using dd method from terminal and everything worked fine when used USB to boot live and then actually install q4os 3.7 to bare metal.
Today I used an old but good image of LXLE 16.04 as a test of Etcher on Lite. Same result of error message but the USB booted live and program ran just fine.

Cylindric, 2019-07-17:
My error comes up as soon as I click start, and nothing is ever written to the card. Doesn't get that far.

DenjW, 2019-07-17:
Just installed two versions of the new Sparky Linux distros to USBs using
Etcher and in both cases got red error message AFTER Flashing was completed
and validation done.
However can boot both versions live on laptop with no error installing or
using live.  Will be installing Xfce version to SSD shortly and don't
expect a problem.
Don't want to see this error message every time I use Etcher.

gtanyware, 2019-09-26:
Getting the same thing, on KDE Neon. I can write the same image to a DVD with no trouble.

lurch, 2019-09-26:
Looks like the relevant output from @Cylindric 's error log is:
`Error: ENOENT, node_modules\\xxhash\build\\Release\\hash.node not found in C:\\Users\\mark\\AppData\\Local\\Programs\balena-etcher\resources\\app.asar`

blakeedwards823, 2019-10-17:
I  am experiencing the same issue since updating my Mac to Catalina. Is there any fix for this problem yet?

lurch, 2019-10-17:
For issues with Catalina, see #2833

DenjW, 2019-10-18:
No problem for me for some time now but Etcher was always working for me
just with incorrect red error message.
Obviously corrected a while ago.  I use Etcher all the time for putting
ISO's on USB drives.

meredithheskethfortescue, 2019-11-04:
> No problem for me for some time now but Etcher was always working for me just with incorrect red error message. Obviously corrected a while ago. I use Etcher all the time for putting ISO's on USB drives.
> […](#)

Funny. I had no problems in the past, but I have this issue now.

Cylindric, 2019-12-04:
FYI still happening with 1.5.66.

evanlandreneau, 2019-12-05:
Happening to me to!

veeshi, 2019-12-14:
Still getting the same issue

sengenbe, 2019-12-18:
Same issue 1.5.68.  Fresh sd

DenjW, 2019-12-18:
Looking at all your comments since July 2019 I realized I was still using
version 1.5.52 2019-07-22 on Peppermint 10 linux. I have not updated
because Balena do not prompt you for updates like Etcher used to do (at
least I do not get update reminders!).  Maybe I'm lucky because I used to
have this problem on an earlier version but no longer and seems obvious
that I will not be updating any time soon (if it ain't broke don't fix
it!).  Sorry can't help you all with current problems unless you can still
get version 1.5.52.  There are alternatives to Etcher but if it works with
just red flag afterwards I would just ignore it.  Good luck all!

lurch, 2019-12-18:
Older versions are at https://github.com/balena-io/etcher/releases

feren, 2020-01-17:
Getting the same issue on Feren OS

eddiechilton, 2020-02-07:
Same issue with 1.5.76 on KDE Neon

awarebayes, 2020-02-07:
same on arch and manjaro

DenjW, 2020-02-07:
I am using 1.5.67 20191206 on Peppermint 10 respin with no problem.
That happened to be the current version when I updated the distro.  I do
not now update Etcher regularly as they are putting out so many new
versions there are increased risks of breakage.  You can get old versions
on Github site.

eddiechilton, 2020-02-10:
My problem was with the AppImage for 1.5.76, so I installed the latest DEB package and it worked. Saw this recommended course of action on [this thread](https://github.com/balena-io/etcher/issues/2859).

LanceHaverkamp, 2020-03-11:
Same problem with KDE Neon (Based on the current Ubuntu LTS).  Etcher stopped working several months (maybe a year) ago.  Writing with dd from the command line works.

PietroJominiITIS, 2020-03-23:
same problem in version `1.5.79`, "solved" by starting the appimage with administrator permissions

huntantr, 2020-05-28:
I've been having the same problem under KDE Neon. I was running 1.5.71 and upgraded to 1.5.94 but still have the same issue. Can't write any ISO's to an USB drive.

thundron, 2020-05-29:
It looks like we ran into everyone adding to different issues in this thread. I'm closing as there's too much confusion.
For everyone still having issues: please open a new github issue and be sure to follow the instructions written in the template so we'll be able to narrow down what's happening in all your cases

closed 2020-05-29 by thundron