balena-io/etcher #2855: Trojan/Win32.GenKryptik Windows 10 x64 Etcher-1.5.51

sumitxbee, 2019-07-16:
[image]
- **Etcher version:** 
- **Operating system and architecture:** 
- **Image flashed:** 
- **Do you see any meaningful error information in the DevTools?** 

Hi, I've downloaded Etcher software from its official website https://www.balena.io/etcher/, thought i'd scan it with Virustotal online tool.. look what I've found "Trojan/Win32.GenKryptik" This is latest version and I'm on windows 10 pro machine, I'm going to switch to linux and used it to burn an iso file. please find attached picture file for details.

deadair3210, 2019-07-16:
Probably a false positive, usually generic detections are just the scanner being over protective

thundron, 2019-07-22:
@sumitxbee Thank you for the report. As @deadair3210 mentioned it's a false positive, we've seen this happen a lot when Etcher first released because security programs didn't trust yet our certificate (as more people install the app, the false positives disappeared), but wrt online tools we might need contact them directly