balena-io/etcher #2856: Add zstd support

sedlund, 2019-07-16:
Please add support for zstd compressed images.

lurch, 2019-07-16:
Also previously discussed in #711

thundron, 2019-07-22:
Thank you for your request. As @lurch pointed out it was previously discussed about a year ago, so we can at least see if the situation got better wrt dealing with 7-zip

lurch, 2019-07-22:
@sedlund Out of curiosity, are you aware of disk images that are being distributed in zstd format?

sedlund, 2019-07-22:
@lurch I haven't seen anything offered as of yet - I suppose its a bit chicken and egg now. 

I use it for creating SD backup images compressed on the device and then streamed over the network to its final destination.  I'm sure you've seen the benchmarks, Zstd is far and away faster in compressing on small devices, 

7zip itself doesn't have Zstd support.  There is a fork that includes it:
https://github.com/mcmilk/7-Zip-zstd/
There are also some good benchmark graphs showing the advantages, although using an older Zstd 1.2 - it has received many large performance increases since then.

I'm not sure why you'd need 7zip to implement Zstd though.  I would think it would be implemented similarly to how lzma was in Etcher.

lurch, 2019-07-22:
> I would think it would be implemented similarly to how lzma was in Etcher.

A quick search finds https://www.npmjs.com/package/zstd which was last published 4 years ago and https://www.npmjs.com/package/node-zstandard which was last published 3 years ago :slightly_frowning_face: 
More chicken & egg? :chicken: :egg:

sedlund, 2019-07-24:
@lurch I guess a lot of iteration is not required when you only need to implement 2 methods to hook to an external lib - compress and decompress eh?  😃

lurch, 2019-07-24:
Sorry, I was indirectly referring to your "using an older Zstd 1.2 - it has received many large performance increases since then" comment - have these "performance increases" been added before or after these node packages were last published? :man_shrugging:

jakogut, 2020-10-08:
I've been looking into this because of the speed of ZSTD's decompression, combined with its good compression ratio, which I think offers a uniquely useful algorithm for compressing disk images and the like.

It looks like the main blocker is the age of existing ZSTD node modules, the lack of a module that implements a node Transform stream, and the small Venn diagram of people who know Node and C++ well enough to make a native module implementing this interface.

Fortunately, compiling libzstd to WebAssembly is pretty straightforward with Emscripten, at which point we can just call exported methods directly from JavaScript. I'm no JS guru, but I know my way around Emscripten, and this is an interesting project to me, so I guess we'll see where it goes.

bergmanu, 2022-03-07:
Is there any news on this topic? 
I would also require to flash .zst compressed images

akrabu, 2022-03-07:
I would greatly appreciate this feature as well!  Thanks!

rpardini, 2022-03-08:
I am producing `.img.zstd` images for Armbian, and would push for that to be standard if Etcher supported it.

robertlipe, 2022-05-07:
Really large images, like operating systems, are starting to use zstd more. That's what brought me here.
https://github.com/starfive-tech/Fedora_on_StarFive

So to @lurch's question "are you aware of disk images that are being distributed in zstd?" the increasing answer is yes and interest seems to be trending up. I hadn't heard of ZSTD before a year ago, so I don't perceive this as a red hot issue, but it must be growing.

In my case, I just wanted a streaming decompress so I don't have to read the big file to write a bigger file and then use Etcher to copy that. I'd be perfectly happy for that 'bigger' image to exist only in temporary memory as it was streamed to the USB card.

While I'm here, thank you for a great cross-platform utility. I know those are hard to make and Etcher rocks.

foundObjects, 2022-06-10:
Chiming in here - I use zstd heavily in my image workflow for SBC distro and ChromeOS images. Zstd support would save me a lot of time, xz is space efficient but has crappy compress/decompress compute and time efficiency.

symgryph, 2022-07-11:
would be nice to add to etcher pro!

mrbluecoat, 2022-08-06:
I agree. Since #2048 was closed as uncompleted, `zstd` is another good option for those who don't want the long wait times and high resource consumption of `xz`

mrbluecoat, 2024-04-29:
Good timing ping. With the recent xz security fiasco, many distros are reevaluating or switching to zstd, such as OpenWrt.

Impact123, 2024-10-08:
The `-T 0` parameter can speed `xz` up tremendously as it seems to default to just a single thread. Until this feature request is implemented this might help. It seems obvious to check/know but I'm sharing it because it took me longer than I'm comfortable to admit to actually look at the process' CPU utilization when I encountered `xz`'s slow (de-)compression.

sedlund, 2025-05-07:
For anyone wanting a GUI program that flashes images and supports zstd check out [Rufus](https://github.com/pbatard/rufus/releases).  It is only a few megabytes in size compared to Etchers now 250MB.

nicomiguelino, 2025-06-05:
I quickly checked v2.1.2 of Etcher and zstd is still not supported.
I tried [Raspberry Pi Imager](https://github.com/raspberrypi/rpi-imager) v1.9.3, but I got the following:

[image]

I think they already fixed the issue (see https://github.com/raspberrypi/rpi-imager/issues/1002 and https://github.com/libarchive/libarchive/issues/2488 for details) in [v1.9.4](https://github.com/raspberrypi/rpi-imager/releases/tag/v1.9.4).

Still waiting for zstd support in Etcher. In the meantime, the Raspberry Pi Imager seems to be a reliable alternative.