balena-io/etcher #2859: error: please check that the archive is not corrupted. every iso i try

roadh0use, 2019-07-21:
- **Etcher version:** 
1.5.51
- **Operating system and architecture:** 
ubuntu 19.04 64bit
- **Image flashed:** 
every manjaro iso i can find
- **Do you see any meaningful error information in the DevTools?** 
i really dont know what im looking at here

every iso i try to flash i get an error saying "something went wrong. if it is a compressed image, please check that the archive is not corrupted. i have tried manjaro architect iso, manjaro i3 iso, manjaro kde iso, etc. every one gives this error.

lurch, 2019-07-21:
@zvin @thundron https://www.gnu.org/software/bash/manual/html_node/Exit-Status.html says `If a command is found but is not executable, the return status is 126.`

thundron, 2019-07-22:
@roadh0use are you using the portable version?

roadh0use, 2019-07-30:
sorry i just had a chance to reply. no i was just using regular iso's. here is a link to the architect download i was using. https://osdn.net/projects/manjaro/storage/architect/18.0.2/manjaro-architect-18.0.2-stable-x86_64.iso/

roadh0use, 2019-07-30:
also just tried with the new 1.5.52 appimage with a different iso. this time it was the manjaro i3 x86_64 iso. 
`/tmp/.mount_balena9GuAtS/resources/app.asar/node_modules/raven-js/dist/raven.js:58 {"stack":"Error: The elevated process died unexpectedly\n    at Object.exports.createError (/tmp/.mount_balena9GuAtS/resources/app.asar/lib/shared/errors.js:253:17)\n    at Object.exports.createUserError (/tmp/.mount_balena9GuAtS/resources/app.asar/lib/shared/errors.js:292:18)\n    at Bluebird.using (/tmp/.mount_balena9GuAtS/resources/app.asar/lib/shared/permissions.js:207:22)","message":"The elevated process died unexpectedly","description":"The process error code was 126","report":false,"image":"manjaro-i3-18.0.4-stable-x86_64.iso"}`

roadh0use, 2019-07-30:
interesting. i removed the newest appimage and tried the newest .deb instead and its working.

thundron, 2019-07-30:
@roadh0use Thank you, I'm gonna fiddle with the latest appimage to see if I can reproduce!

vgaleksa, 2019-08-03:
Dont use appimage, install etcher package with pacman, or try installing gconf and using appimage.
I had same problem and then checked for etcher package, installed it and saw that it installed gconf also and now etcher works.

thundron, 2019-08-14:
@roadh0use Did you try @vgaleksa 's suggestion?

iMerica, 2019-10-11:
I had the same problem. The solution was to unmount all the extra volumes.

`diskutil list`

```
/dev/disk2 (external, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:     FDisk_partition_scheme                        *31.3 GB    disk2
   1:             Windows_FAT_32 NO NAME                 31.3 GB    disk2s1
```

`diskutil unmount disk2s1`

Etcher should work now. 

Optional:

Skip etcher and just use `dd`:

`sudo dd if=/path/to/your/disk.img of=/dev/rdisk2 bs=1m`
`diskutil eject /dev/disk2`

lurch, 2019-10-11:
> The solution was to unmount all the extra volumes.

AFAIK, Etcher _should_ be doing that automatically?

lurch, 2019-11-15:
@pczekalski Can you check if the AppImage still fails _after_ you've installed the DEB package?

pczekalski, 2019-11-15:
Yep - I'm home, away of TF cards I've been using while doing the flashing that failed, but here, using the same OS and PC but different, 128GB TF card (previously experimented with 8GB and 16GB)  I can acknowledge that once DEB was installed, appimage works as expected (no error message). 
If the size of the TF card is meaningful then let me know, I can test it earliest next Monday.

zingbretsen, 2019-12-07:
Same issue with the appimage on Ubuntu 19.10. It works when I install the DEB package and use that. The appimage version still does not work after installing the DEB package.

rotimi-best, 2019-12-07:
> Same issue with the appimage on Ubuntu 19.10. It works when I install the DEB package and use that. The appimage version still does not work after installing the DEB package.

This was what helped me. Instead of using the Appimage, I had to download the package using `apt-get` as stated in the README and it is working fine.

garethtdavies, 2019-12-09:
I've reproduced this on Ubuntu 19.10 and can confirm that the AppImage versions >= 1.5.66 have this issue so likely introduced in 1.5.66 as successfully tested on 1.5.65 and lower.

The current deb as noted above works.

jhoughjr, 2019-12-14:
using etcher on Mac OS X Catalina. Also every image I burn it says is corrupted.

jtagcat, 2020-01-09:
Fedora 31, same issue, with known-good isos written before on same machine, known-good usb-sticks-

AeonFox, 2020-01-13:
Have you tried downloading the newest BalenaEtcher? That fixed my issue! Hope this helps.

NickJorens, 2020-01-22:
I still have this issue happening on the latest version as of writing. On the AppImage that is. I installed it through DEB and it's working again! :)

huahuayu, 2020-03-12:
> I had the same problem. The solution was to unmount all the extra volumes.
>
> `diskutil list`
>
> ```
> /dev/disk2 (external, physical):
>    #:                       TYPE NAME                    SIZE [...]

hi, why it's rdisk2 in `of=/dev/rdisk2` instead of disk2

huahuayu, 2020-03-12:
I download etcher in https://www.balena.io/etcher/ today, for mac the version is 1.5.45, but in github release page, the latest release is 1.5.79, so, the website download link should be updated.

zvin, 2020-03-12:
@huahuayu the website fetches the latest release from github ( from here: https://raw.githubusercontent.com/balena-io/etcher/gh-pages/scrutinizer.json ).
Something in your browser is blocking this request.

lurch, 2020-03-12:
> hi, why it's rdisk2 in of=/dev/rdisk2 instead of disk2

See #1042

tthebst, 2020-04-17:
I had this issue as well on Mac OS. I solved it by running etcher from the terminal with following command:
` sudo /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher `

vincegre, 2020-05-13:
Used the deb package instead the AppImage and everything works fine :+1:

wzJun1, 2020-09-19:
> I had this issue as well on Mac OS. I solved it by running etcher from the terminal with following command:
> `sudo /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher`

this is the correct solution

gimdh, 2020-10-05:
I'm on Ethcer 1.5.109, and it seems the error has come back with Catalina 10.15.7 update.
I want to point out that the error message should be fixed as well as wrong behavior, since it's quite misleading.

Sidordroid, 2020-10-12:
What's wrong with this balenaEtcher is not one thing like that.
[image]

thundron, 2020-10-13:
@Sidordroid that looks like a separate issue than the one descripted above

pngimbwa, 2020-11-06:
To solve the problem, just right-click the software (i.e. balenaEtcher) and run it as administrator on Windows PC.

cautious1, 2020-11-06:
To parrot pngimbwa,  run Etcher as administrator.

kyxap, 2020-11-12:
For Mac users (catalina): to resolve this issue just install app with `brew` like:
```bash
brew install balenaEtcher
```

and run app from Application folder without sudo\admin\etc

lborrazas, 2020-12-30:
Two guys said this before, but ill say it once again cause it is important.

If your error message says "PERMISSION DENIED"
Then just running Balena As An Administrator solves everything :D

NavadeepGaneshU, 2021-06-02:
I got the same issue while using balena etcher while used when it opened by default after new installation. Closed that and reopened the software as administrator upon which it works very well.

closed 2021-10-19 by thundron

ethaniel, 2021-11-01:
Had this problem on my macbook, updated Etcher to 1.5.122 - problem is gone.