balena-io/etcher #2860: The writter ended unexpectedly. Please try again...etc

MaXiNom1, 2019-07-22:
Etcher version: 1.5.51
Operating system and architecture: Mac osx High Sierra 10.13.6
Image flashed: 2019-07-10-raspbian-buster-full.zip 
Do you see any meaningful error information in the DevTools?: None that I can see

yslupdates, 2019-07-22:
Having same issue with MacOS Catalina beta and trying to flash dietpi

MaXiNom1, 2019-07-23:
Thanks, ended up using dd tool in terminal - honestly a faster option anyways as long as you are familiar with the risks and are comfortable with terminal. (dont flash your main)

yjhmelody, 2019-08-11:
Why my sd card could not be recognized after getting such a error ?

lurch, 2019-08-12:
@yjhmelody see https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#recovering-broken-drives

yjhmelody, 2019-08-12:
@lurch  It do not work well. There are nothing changed.

yjhmelody, 2019-08-12:
Well, I found diskpart's example and finally resolved it. Thanks!

thundron, 2019-09-25:
@yslupdates For Catalina, see https://github.com/balena-io/etcher/issues/2833#issuecomment-511152047
@MaXiNom1 Strange to see this error on OSX without any context in the logs. Feel free to reopen this issue if you ever use Etcher and see this pop up again! Usually, there's more insight on the logs *above* the error message in the DevTools Console tab (`cmd + shift + i` on OSX)

closed 2019-09-25 by thundron