balena-io/etcher #2865: The writer process ended unexpectedly.

dougabugg, 2019-07-26:
- Etcher 1.5.51
- Windows 10 x64
- tails-amd64-3.15.img
- See attachment
[etcher.log](https://github.com/balena-io/etcher/files/3434742/etcher.log)

dougabugg, 2019-07-26:
I have an update. I had previously installed Fedora on the USB drive using Fedora Media Writer 4.1.4-1.fc30, and found that I could not manipulate the USB drive in Windows's Disk Management console (formatting partitions specifically would fail). I opened the USB drive in Fedora Media Writer to see if that could still write the USB drive, and found that FMW displayed a popup with an option to "Restore" the USB drive and remove the Fedora live image I had written previously. After letting FMW restore the USB drive, Etcher worked flawlessly.

thundron, 2019-07-29:
Thank you for your report and investigation, from the logs it looks like there was an issue with the `diskpart` command; the reason is probably because of some sectors not being healthy, as you found out.

closed 2019-07-29 by thundron