balena-io/etcher #2866: When I launch Etcher, I get a blank screen.

inismor, 2019-07-26:
- **Etcher version:** 
1.5.51 (1.5.51)

- **Operating system and architecture:** 
macOS Mojave 10.14.5

- **Image flashed:** 
Never got that far

- **Do you see any meaningful error information in the DevTools?** 
"DevTools was disconnected from the page. Once page is reloaded, DevTools will automatically reconnect."

[image]

lurch, 2019-07-29:
Similar to #2659 ?

DouglasHeriot, 2019-08-01:
I have the same issue on Windows 10 (latest 1903)
I installed Etcher fresh from the website.

crackingthecodethomas, 2019-08-02:
Same here. Shows the normal grey etcher screen and then goes white where i cant click anything. Ctrl + Shift + I says Dev tools was disconnected.

thundron, 2019-08-14:
Could you try opening the devtools then reloading by pressing `ctrl + R` ? That *should* show the content of the previous logs too. If not, try enabling logs persistence in the console settings

inismor, 2019-08-14:
thundron, I did as you asked. It goes off for a second and then returns with the same error. I've turned on log preservation. This is the response I get in the console:
```
Navigated to file:///Applications/balenaEtcher.app/Contents/Resources/app.asar/lib/gui/app/index.html
/Applications/balenaEtcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:58 Download the React DevTools for a better development experience: https://fb.me/react-devtools
You might need to use a local HTTP server (instead of file://): https://fb.me/react-devtools-faq
/Applications/balenaEtcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:58  _____ _       _
|  ___| |     | |
| |__ | |_ ___| |__   ___ _ __
|  __|| __/ __| '_ \ / _ \ '__|
| |___| || (__| | | |  __/ |
\____/ \__\___|_| |_|\___|_|

Interested in joining the Etcher team?
Drop us a line at join+etcher@balena.io

Version = 1.5.53, Type = dmg
/Applications/balenaEtcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:58 Wed Aug 14 2019 17:17:35 GMT+0100 (BST) Application start ({"packageType":"dmg","version":"1.5.53","applicationSessionUuid":"43a54a92-2272-4ff1-bb1d-81a5e5bb2481","sample":0.1})
```

and then the alert pops up about DevTools being disconnected. All of the while, every window is just white. Oddly, when I trace the URL, `app.asar` turns out to be a document, not a directory, so I'm not sure how it was supposed to work.

lurch, 2019-08-14:
You can read more about `.asar` files at https://github.com/electron/asar

DavidFW1960, 2019-12-10:
same issue here

pawankumar30, 2019-12-25:
Try Running with low configuration under compatibility, also with reduces resolution. It worked for me at 640*480 resolution.

quamis, 2020-04-12:
Had the problem, on Ubuntu, devtools (ctrl+shit+i) reported the same error as here: https://github.com/signalapp/Signal-Desktop/issues/2641

Fixed as suggested in the above issue by changing /tmp:

TMPDIR=./tmp ./balenaEtcher-1.5.80-x64.AppImage 

As a note, running things from /tmp seems like a security issue.

lurch, 2020-04-14:
@quamis Hmm, maybe something like #1699 is affecting Etcher again? :confused:

quamis, 2020-04-14:
Yeah, seems so. Using a non default temp folder worked as expected

stanleeway, 2020-12-04:
to fix this on windows 7 run it at 256 colors,it should work

Stefanious, 2022-12-09:
Same issue, Zorin, Ubuntu, Debian live disk... Does not work. Shame. Switched to Ventoy...

glaubersm, 2023-03-19:
Same problem with balenaEtcher-Portable-1.18.4 on Windows 7 Professional 64 bits.

jimboelter, 2023-04-21:
Same, version 1.18.4 on Win7 Pro 64, just ran Rufus and built my boot image

closed 2023-07-12 by dfunckt