balena-io/etcher #2870: Provide packages for ARM architectures

imrehg, 2019-07-29:
Should provide packages which can be used e.g. on Raspbian to burn image when running on a Raspberry Pi. Currently should be possible to build from source, but packages would make things a lot easier for users.

lurch, 2019-07-29:
Duplicate of #2748 ? (amongst other similar requests)

jpie726, 2019-11-02:
Any progress yet?

Botspot, 2020-01-09:
One user on the Pi forums has compiled an [Etcher armhf deb]Etcher .deb for Raspberry Pi(https://www.raspberrypi.org/forums/viewtopic.php?f=62&t=255205&start=25#p1564958).
Installation is easy:
`wget https://github.com/futurejones/balena-etcher-arm/releases/download/v1.5.63/balena-etcher-electron_1.5.63_Raspbian_Buster_armhf.deb`
`sudo apt-get install ./balena-etcher-electron_1.5.63_Raspbian_Buster_armhf.deb`

kkarhan, 2021-12-16:
Any developments on this issue?

After all it should just be another build target in the existing pipeline...
https://github.com/balena-io/etcher/issues/2870#issuecomment-572798550

Botspot, 2021-12-16:
[This user on github](https://github.com/Itai-Nelken/Etcher-arm-32-64) has compiled Etcher debs and has an [automated script](https://github.com/Itai-Nelken/Etcher-arm-32-64) to do so. It would be great if this was made official!

theofficialgman, 2022-12-15:
@Botspot please delete/edit your posts here that refer to balena-etcher-electron.
it is being referenced in other posts and users who don't look hard will end up with this bad out of date package that doesn't match the upstream naming scheme

Botspot, 2022-12-15:
Done.

theofficialgman, 2023-12-21:
@dfunckt issue should be reopened. No armhf and arm64 Linux packages are being produced

dfunckt, 2023-12-22:
You are right of course, in my haste I mistook the issue to be about Apple Silicon builds and didn’t read the discussion.

I spend a lot of time trying to get cross compilation going but couldn’t and I had to move on. I haven’t given up though, will give it another go as soon as I get a chance.

Kha1r0, 2024-05-14:
Any updates on this?

Abhimanyu14, 2024-10-25:
I am using `Etcher for macOS (arm64)` provided [here](https://etcher.balena.io/#download-etcher) on a Mac Mini with M2. 

I format the USB using GUID, but after flashing using Etcher, it changes to Master Boot Record (MBR), and the USB is not shown in the Startup screen.

Kha1r0, 2024-10-25:
> I am using `Etcher for macOS (arm64)` provided [here](https://etcher.balena.io/#download-etcher) on a Mac Mini with M2.
> 
> I format the USB using GUID, but after flashing using Etcher, it changes to Master Boot Record (MBR), and the USB is not shown in the Startup screen.

This is due to how ISOs work. The ISO file you're using to flash your USB drive contains information about the partition table and file system to use. So your ISO probably just says to use MBR and a file system macOS can't read.

But thats completely besides the point of this thread. The macOS Arm package is here, so this is more or less only about Windows and Linux now.

Abhimanyu14, 2024-10-25:
Hi @Kha1r0,
I am using Ubuntu ARM ISO - https://ubuntu.com/download/server/arm.

If this is not the correct place, can you please let me know what I should do or where I should check further on this?
Thank You.

Kha1r0, 2024-10-25:
It seems your USB Drive is just ready to be used. Is it not working?

Abhimanyu14, 2024-10-25:
I see these after formatting/erasing my USB device.
[image]
[image]

Etcher from here - "Etcher for macOS (arm64)"
https://etcher.balena.io/#download-etcher

Ubuntu from here - https://ubuntu.com/download/server/arm

Kha1r0, 2024-10-25:
Oh. You‘re trying to install Linux on your apple silicon mac. You have to use https://asahilinux.org/ to do that. Please inform yourself thoroughly as this is not an easy process. But as i said, we‘re completely off the topic of this issue.

Sorry @imrehg for congesting your issue.

PR #4132 (2023-12-21): Modernize build pipeline
Switch to electron-forge which is now officially adopted by the Electron project. Solves a bunch of issues along the way.

Issues related to experimental v1.18.13:

Closes #4138 
Closes #4129
Closes #4119 

Build related issues:

Closes #4136 
Closes #4125 
Closes #4103 
Closes #4087 
Closes #3767 
Closes #3728 
Closes #3653 
Closes #3400 
Closes #3292 
Closes #3269 
Closes #3263 
Closes #2870 
Closes #2826 

(Admittedly, some of the issues are linked opportunistically -- please reopen them if the issue is not actually solved)