balena-io/etcher #2871: portable version creating directories outside of main folder

githubaccount11111, 2019-07-29:
The ones I've found (so far) are balenaEtcher and balena-etcher - both inside the appdata/roaming folder. A portable app should not create anything outside of main directory, that defeats the whole purpose. One of the main reasons I opt for portable software is to keep my system stable and cruft free. Either stop misleading people and call the non-portable (which would be silly since there already is a non-portable version), remove it altogether or make it actually portable, i.e. self-contained and sandboxed.

- **Etcher version:** 1.5.52
- **Operating system and architecture:** Windows 1809 64
- **Image flashed:** yunohost-stretch-3.5.2.2-amd64-stable.iso
- **Do you see any meaningful error information in the DevTools?**

thundron, 2019-07-31:
Can you paste here the content of both folders? It would be easier for us to narrow down what are the files that aren't self-contained, but I can tell right away that one of them should be the `config.json`

githubaccount11111, 2019-08-06:
the balenaEtcher folder has an empty logs folder. Still shouldn't be there in the first place.
the balena-etcher folder has a ton of files and folders which in turn have a ton of files an folders. webrtc, patitions, blobs etc. One of them is indeed config.json. I can't list them all there's just so many of them.

now if that doesn't seem bad enough, I've later realized that Etcher seems to have destroyed my flash drive. Or at the very least messed it up. 16GB magically turned into 369MB which is lovely.

why have I never experienced any of this with rufus?

lurch, 2019-08-06:
> I've later realized that Etcher seems to have destroyed my flash drive.

#2102

lurch, 2019-08-06:
@thundron If you want a "temporary but clean" version of Windows (which I guess will be useful for tracking down what these extra files are) see https://developer.microsoft.com/en-us/microsoft-edge/tools/vms/

githubaccount11111, 2019-08-07:
Fixed the flash drive issue by formatting w/ rufus. Also flashed the same image right after (again w/ rufus), and now the storage is correctly identified and is actually still readable in Windows.
All the comments about the file system and the extra space being ignored are simply false.
Sure, the drive wasn't broken, but it sure as hell was an inconvenience and cost me a few nerve cells.

lurch, 2019-08-07:
> All the comments about the file system and the extra space being ignored are simply false.

No they're not. The difference is that Rufus works in a different way to Etcher. See #413

Rufus and Unetbootin only work with specific images, Etcher works with **any** disk image.

thundron, 2019-08-14:
We're still looking for the files that aren't self-contained, but if you mention that there are more than the basic `config.json`, I feel that's out of our control and it's how electron packaged apps work, at least with the packager we use (which is the most popular, to my knowledge). We'll try to reduce them anyways but I don't think we'll ever have a completely self-contained application, as it needs to access resources out of its scope too. We'll re-open this issue shall we find otherwise.

As for the OT: Some images are formatted in a way that Windows cannot understand, often leading users to think the target has "shrinked" to a lower size. That happens because Etcher writes byte-to-byte whatever the image is, without applying any logic as opposed to what Rufus does instead, and that's to ensure a 1-to-1 reproduction of the flashed image that works cross-platform. Otherwise we'd need to insert more and more OS-specific stuff that we don't want to (as of now, at least)

closed 2019-08-14 by thundron