balena-io/etcher #2876: Erro in install ecther on arm64 in pi

SureshKumar311, 2019-08-04:
- **Etcher version:** balena-etcher-electron_1.5.52_amd64.deb
 **Operating system and architecture:**  ubuntu-mate-18.04.2-beta1-desktop-arm64+raspi3-ext4.img.xz
 **Image flashed:** same 
- **Do you see any meaningful error information in the DevTools?** Worng architecture "arm64"  -run dpkg add- architecture  to add it and upgrade after wards

BopItFreak, 2019-08-04:
I used another version of etcher on raspberry pi 3, and it also didn't work. I got exec format error. Not sure what version is correct.

thundron, 2019-08-05:
We don't currently distribute packages for `arm64` arch. You can try building from source following the guide @ https://github.com/balena-io/etcher/blob/master/docs/CONTRIBUTING.md#developing and then running `make electron-build`

closed 2019-08-05 by thundron

lurch, 2019-08-05:
See #2870