balena-io/etcher #2877: microSD Card becomes read only after flashing with etcher

rg337, 2019-08-05:
Etcher version: 1.5.51
Operating system and architecture: Windows 10, 64 bit
Image flashed: CentOS-Userland-7-armv7hl-RaspberryPI-Minimal-1810-sda.raw.xz
Do you see any meaningful error information in the DevTools? No

I have a supposedly 16GB microSD card.

OS: Windows 10
Downloaded balena Etcher Portable: https://github.com/balena-io/etcher/releases/download/v1.5.51/balenaEtcher-Portable-1.5.51.exe
Downloaded CentOS Raspberry image: https://mirror.xtom.com.hk/centos-altarch/7.6.1810/isos/armhfp/CentOS-Userland-7-armv7hl-RaspberryPI-Minimal-1810-sda.raw.xz

After flashing the image, I have SD card with 3 partitions:
Details as below:-

====================================================================================================
DISKPART> list part

There is no disk selected to list partitions.
Select a disk and try again.

DISKPART> list disk

  Disk ###  Status         Size     Free     Dyn  Gpt
  --------  -------------  -------  -------  ---  ---
  Disk 0    Online          476 GB      0 B        *
  Disk 1    Online           14 GB    12 GB

DISKPART> select disk 1

Disk 1 is now the selected disk.

DISKPART> list part

  Partition ###  Type              Size     Offset
  -------------  ----------------  -------  -------
  Partition 1    Primary            668 MB  1024 KB
  Partition 2    Primary            488 MB   669 MB
  Partition 3    Primary           1430 MB  1157 MB

DISKPART> clean

DiskPart succeeded in cleaning the disk.

DISKPART> create partition primary

Virtual Disk Service error:
The operation timed out.

DiskPart has referenced an object which is not up-to-date.
Refresh the object by using the RESCAN command.
If the problem persists exit DiskPart, then restart DiskPart or restart the
computer.

DISKPART> rescan

Please wait while DiskPart scans your configuration...

DiskPart has finished scanning your configuration.

DISKPART> create partition primary

Virtual Disk Service error:
The operation timed out.

DiskPart has referenced an object which is not up-to-date.
Refresh the object by using the RESCAN command.
If the problem persists exit DiskPart, then restart DiskPart or restart the
computer.
====================================================================================================

My card now kind of becomes read-only. I am not able to clean, format, delete files or do anything now.
Kindly let me know how can I reclaim my card.

I have tried using the below software, but not able to format my card :(

AOMEI Partition Assistant Standard Edition 8.3
http://www2.aomeisoftware.com/download/pa/PAssist_Std.exe
Tried to format a partition. :(
Tried to delete a partition. :(
Tried to delete all partitions on disk. :(
Moreover message for all above operations is, The operation is completed and all operations are successful.

EaseUS Partition Master 13.5
http://download3.easeus.com/free/epm.exe
Tried to format a partition. :(
Tried to delete a partition. :(
Tried to delete all partitions on disk. :(
Moreover message for all above operations is, The operation is completed and all operations are successful.

HDD LLF Low Level Format Tool
https://hddguru.com/software/HDD-LLF-Low-Level-Format-Tool/HDDLLF.4.40.exe
Tried quick wipe as other mode gives Format Error at some sectors.
Program hanging in between. Have to detach the card to kill the program.	

IM-Magic Partition Resizer Free
https://download.resize-c.com/resizer-free.zip
Format. As per message "1 operation has been executed successfully". But no change on SD card. Format not done. :(

SD Memory Card Formatter 5.0.1 for SD/SDHC/SDXC
https://www.sdcard.org/downloads/formatter/eula_windows/SDCardFormatterv5_WinEN.zip
Option for "CHS format size adjustment" not available to enable.
Quick format, Overwrite format. Both Failed with message "Formatting failed." :(

jagp3, 2019-08-05:
have the same issue with a 16 GB micro SD card. I used to flash Linux mint to install in other computer. Now I want to use it for other purposes I just have a read only flash memory with 14KB free of 2.25 MB. If I see it to Windows Management tools for Storage, I see three partitions. However, even when deleting the volumes, I can not delete allt three, I can not merge or extend the other two partitions. etcher does not have a workaround to revert the flashing. Not sure if flashing some windows image will change it

thundron, 2019-08-12:
The size you see is what Windows is able to mount and read, which in the case of flashed linux images, isn't gonna be the full size of the target (sometimes, no partitions at all)
We suggest [following these steps](https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#recovering-broken-drives) in order to recover a drive, but there are other methods you can use too

closed 2019-08-12 by thundron

rg337, 2019-08-13:
The recommendation given above for Windows is to use diskpart command. 
In the problem report above, I have already mentioned about using the same and also the output of diskpart command. Its not helping me to recover the drive.

jagp3, 2019-08-14:
@thundron as @rg337 mentioned. Diskpart was not helpful.

As I mentioned before, I tried to flash a windows image with the app and I got a message with warning about needing another software to make a bootable drive for a Windows installation. I was afraid that my micro SD was gone and just useful for flashing linux images. 

in Windows, I opened Rufus App, left values for partition scheme as it was. The micro SD was recognized by the app as it was listed. Down below, I left the check mark on "Create a bootable disk using" I chose "FreeDos"

Hit Start, and the program formated the micro SD and turned int a USB drive fully usable with the right capacity. Then I went to "This computer" and from that screen I right clicked that drive, went to "Format" again but in the context menu, Hit default values and apply quick format. 
BOOM, it was recognized again as a SD card. hope this helps.

thundron, 2019-08-14:
@rg337 Try formatting with Rufus first as suggested by @jagp3 . Diskpart can sometimes be unreliable for reasons sometimes unknown and as I previously mentioned you can try other recovery methods/applications if diskpart doesn't do the job for you

RyanNgCT, 2020-03-09:
So there is no way that we can write anything after the SD card has been flashed? Because I'm thinking of doing a headless Raspberry Pi 4 setup so I need to add the ssh and other config files or do I do it before I flash the SD card?

Thanks for your help!

jagp3, 2020-03-09:
You can recover your card using rufus in Windows. or Disks utility in any
Ubuntu based Distribution! You just need to reset all the partition scheme.
In rufus, it will detect the drive with original capacity. You can do a Non
bootable drive with it and it will flash it to default file system.

ivan1970, 2021-01-04:
Hello,
I have the same issue after flashed an image with etcher the microsd became read only.
I tried all solution that I found on internet
1) Using diskpart to remove readonly attribute;
2) removing Write Protection into StorageDevicePolicies of windows registry;
3) using Rufus on windows and all other disk utility on Linux Ubuntu and on MacOS

could you help me to understand how to recover my micro sd card?
thanks
Ivan

br00haha, 2023-01-25:
I made an account here just so I could post this since I've had this problem for a few days even though this post is 4 years old.   Was trying Ubuntu 22, it wouldn't install correctly with Balena, wanted to reimage my USB and found that I couldn't do shit with it. Yada yada yada, whatever Balena does I think is geared mostly toward Linux distros. Windows stuff wouldn't work. 

Tl:dr I plugged my USB into my Android figuring they were similar enough. Formatted it on there and now the stick is back to normal. I'm slapping myself at how simple that turned out.

rg337, 2023-01-28:
Thanks for the reply, but now I have misplaced that SD card. Looks fairly simple if I had tried on linux environment.
But appreciate your effort for this post. Hope this post is helpful for anyone who encounters the same in future.