balena-io/etcher #2881: [SOLVED] Error flashing SD card using etcher AppImage on manjaro linux

Niceplace, 2019-08-12:
- **Etcher version:** 
  - 1.5.53 (AppImage) 
  - https://github.com/balena-io/etcher/releases/download/v1.5.53/balena-etcher-electron-1.5.53-linux-x64.zip

- **Operating system and architecture:** 
  - Manjaro Linux (x64)
  - (uname -a) `Linux simon-tp 4.19.56-1-MANJARO #1 SMP PREEMPT Wed Jun 26 03:35:21 UTC 2019 x86_64 GNU/Linux`
- **Image flashed:** 
  - https://osdn.net/projects/manjaro-arm/storage/rpi3/lxqt/19.06/Manjaro-ARM-lxqt-rpi3-19.06.img.xz
- **Do you see any meaningful error information in the DevTools?** 
  - Yes, see issue detail below

## Summary (includes solution): 

When trying to flash an SD card using the AppImage packaged version of Etcher, running on Manjaro Linux, the process fails with a generic error message. Opening the DevTools console gives more details into the error and suggests this is a permissions issue`

```
"Command failed: cd "/home/simon"; "/usr/bin/pkexec" --disable-internal-agent /bin/bash -c "echo SUDOPROMPT; sh '/tmp/tmp-1683OiD12Fzl2VQu.cmd'"
/tmp/tmp-1683OiD12Fzl2VQu.cmd: line 64: /run/user/1000/appimagelauncherfs/0000.AppImage: Permission denied
"
```

**The issue is solved** by deleting the package `appimagelauncher` and rebooting the system (not sure if the reboot is really necessary, but I did it and it worked). Others might need to also remove `libappimage` (from extra)  but that was not the case for me.

Referenced forum solution: https://forum.manjaro.org/t/solved-appimage-woes-when-building-package/86654/6

## Steps to reproduce:

Note: You need to have DevTools open in etcher before doing these steps in order to see the logs (CTRL+SHIFT+I)

1. Ensure `community/appimagelauncher` is installed (the issue could arise because of `extra/libappimage` as well, given the forum post that I read but it was not installed in my case
1.1 To verify if one (or both) of these packages is installed, run `sudo pacman -Ss appimage`
2. Follow the steps to flash an external device in etcher's UI with any valid combination of external drive & image (you can use the image I linked in this post)
3. Notice the following error:

[image]

The devtools shows the following stacktrace: 

```
"Error: Command failed: cd "/home/simon"; "/usr/bin/pkexec" --disable-internal-agent /bin/bash -c "echo SUDOPROMPT; sh '/tmp/tmp-1683OiD12Fzl2VQu.cmd'"
/tmp/tmp-1683OiD12Fzl2VQu.cmd: line 64: /run/user/1000/appimagelauncherfs/0000.AppImage: Permission denied

    at ChildProcess.exithandler (child_process.js:303:12)
    at ChildProcess.emit (events.js:182:13)
    at maybeClose (internal/child_process.js:961:16)
    at Socket.stream.socket.on (internal/child_process.js:380:11)
    at Socket.emit (events.js:182:13)
    at Pipe._handle.close [as _onclose] (net.js:596:12)"
```
I didn't push further to understand exactly what was going on but I thought I'd document it for others because finding out the solution was a very frustrating process for me.

Cheers !

Niceplace, 2019-08-12:
Since this is solved I believe it can be closed but I'll leave it open for a few days in case people need to see it.

thundron, 2019-08-14:
Thanks for the report, very detailed and useful. Closing, will reference if other users encounter the same issue

closed 2019-08-14 by thundron

endorama, 2021-04-24:
I had the same issue today, and solved as described.

I reinstalled `appimagelauncher` (version `2.2.0`) afterwards and is now working as expected even with it installed. `balenaEtcher` version `1.5.116`.

cmfcruz, 2021-05-28:
A user in [this forum thread](https://forums.balena.io/t/etcher-not-flashing-iso/171754) suggests that the etcher package from pacman works for him.

Ghanajoke, 2022-09-30:
file:///home/Hassan/Downloads/%01%90F@@%E8%D7%02@8%08@
when i run etcher, a file appears with 'invalid encoding' and nothing happens.

AKopytenko, 2025-03-22:
I have the same problem today. BalenaEtcher validates after recording the distribution and returns an error ("something went wrong"...). The links to the solutions from the posts above do not work, the topic on the forum is dead. It's a pity that no one wrote here what the solution was!