balena-io/etcher #2887: Catalina beta 6 : Cannot launch Etcher

samouds, 2019-08-20:
- **Etcher version: 1.5.56** 
- **Operating system and architecture: MacOS Catalina beta 6** 

When I try to launch Etcher fro mterminal I get this error : 

```
2019-08-20 22:55:26.561 balenaEtcher Helper[2901:79781] *** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '*** -[__NSDictionaryM setObject:forKeyedSubscript:]: key cannot be nil'
```

lurch, 2019-08-20:
See also #2833 & #2843

samouds, 2019-08-20:
It was working under Catalina beta 5 but stop working since beta 6

thundron, 2019-09-18:
Try giving a go at [this suggestion](https://github.com/balena-io/etcher/issues/2833#issuecomment-511152047)
Closing in favor of the other issue for tracking Catalina beta, we'll probably get on it after the release and we already have plans to update our build process to fix some issues with next OSX version(s)

closed 2019-09-18 by thundron