balena-io/etcher #2888: Target installation

tim-lebedkov, 2019-08-23:
- **Etcher version:** 1.5.56
- **Operating system and architecture:** Windows 10, x86
- **Image flashed:** none
- **Do you see any meaningful error information in the DevTools?** no

The Etcher installer does not respect the /D NSIS option. The installation location cannot be changed.

Command:
"%npackd_package_binary%" /S /D=%CD% && del /f /q "%npackd_package_binary%"

lurch, 2019-08-27:
Duplicate of #1659 and #2793 ?

tim-lebedkov, 2019-08-27:
almost. This one is about silent installation.

thundron, 2019-08-28:
That's how `electron-builder`'s one-click installation works, so this can be considered about customising the installation directory. See [here](https://github.com/electron-userland/electron-builder/issues/596) for reference.
Closing as duplicate of https://github.com/balena-io/etcher/issues/2793

closed 2019-08-28 by thundron