balena-io/etcher #2889: Win10 USB64GB formatted to NTFS but Assigning Primary Partition got erase by etcher

kamae, 2019-08-24:
- **Etcher version:** 
- **Operating system and architecture:** 
- **Image flashed:** 
- **Do you see any meaningful error information in the DevTools?**

kamae, 2019-08-24:
Win10 USB64GB formatted to NTFS using Disk Management (as admin) successfully.  But when I flas a Centos ISO emage using balenaEtcher-portable or setup, flash instantely rease the partition. Otherwise the USB is recognized and I can copy to the USB and run Win10 applications. Security is set to "full control" to Everyone. I am trying to make a bootable CENTOS 7 imge from an ISO image file, either CentOS-7-x86_64-DVD-1810 or CentOS-7-x86_64-Minimal-1810. These files are labeled as "Enhandled Extension" in the Explorer. I need some advice. Thank you.
[image]
[image]
[image]

lurch, 2019-08-27:
>  But when I flas a Centos ISO emage using balenaEtcher-portable or setup, flash instantely rease the partition.

Yup, that's exactly what Etcher is designed to do - replace _whatever_ is already on the drive you've selected with the contents of the image/ISO file you selected. The image/ISO file will already contain  disk-partitions, and these get copied as-is to the destination drive.

> I need some advice.

I did a quick search and found https://linuxize.com/post/how-to-create-a-bootable-centos-7-usb-stick/

After flashing has finished, Windows _may_ be unable to recognise any of the partitions on the drive (due to them now using a Linux filesystem), but the USB disk should still be bootable from your BIOS. https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#flashing-ubuntu-isos

closed 2019-11-06 by zvin