balena-io/etcher #2890: Setting for offline mode (don't transmit any data)

rradar, 2019-08-25:
Would be great if etcher could be used offline.

Actually it can be used offline already. But it would be great if etcher would have setting to don't tell **x companies** in **y countries** that I flash a image right no with etcher.

This is some related to #2766 and #2599 or this https://forums.balena.io/t/serious-privacy-concerns-with-etcher-1-4-4 for example.

The setting could be called: 

> Comply with EU Laws

rradar, 2019-08-25:
It's actually very annoying that I need to unplug/disconnect my client from network/internet to flash a image privately...

thefaj, 2019-09-16:
Agreed…more than just comply with GDPR, be an ethical developer. Don't sell out your users to these awful surveillance companies just because you're curious about how frequently your GUI wrapper around "dd" is getting used.

petrosagg, 2019-09-16:
@rradar @thefaj Etcher already supports this by turning data collection off in the settings panel. There is a bug with the data collection libraries that results in a single HTTP request when you `require()` the library. We're fixing this problem by only requiring the dependency if the checkbox in the settings page is checked. This will be released in the following days.

Is there something else this issue is referring to?

thefaj, 2019-09-16:
Opt in…don't bury an setting to opt out in a Settings panel. When the app starts up for the first time, it sends a ton of data back to these third parties.

petrosagg, 2019-09-16:
The opt-in discussion is happening in #2766. I'll keep this issue open for a while in case @rradar meant something other than the setting that already exists.

rradar, 2019-09-20:
@petrosagg etcher transfers a hell lot of (unnecessary and probably illegal in Europe) data to various company's over existing network connections. Best is to call this a bug (your lawyers will tell you same)...

Problem is there is **NO reason for the user to give etcher any access to the internet at all**. If I turn my network connection physically off (turning wifi off and/or unplug my ethernet) etcher will just burn the image and verify it like it should. Same is using etcher witch **little snitch** or **open snitch** (app based firewall) and block all network access and therefor block all unauthorized calls.

If I stay connected and trust etcher it will just send out information (including personal identifying information) to various company's I didn't make any contract with. This is just not acceptable.
So this issue is really about to don't transmit any data. Not to balena, not to google, not to anyone. All data belongs to me, all data stays on my device. Easy? easy!

thefaj, 2019-09-20:
Agreed. Too many developers assume that surveillance is acceptable. “Ethics” needs to be a requirement in computer science courses. There is no acceptable reason to have this malware in the app. (And how does somebody “accidentally” include a third party library?)

petrosagg, 2019-09-20:
> @petrosagg etcher transfers a hell lot of (unnecessary and probably illegal in Europe) data to various company's over existing network connections. Best is to call this a bug (your lawyers will tell you same)...

There we go again with the illegal claims. No, not all data transfers are illegal, you're wrong on that. We are in fact (including me personally) talking with our lawyers and that's not the case. We're spending time money and effort to ensure that we are compliant with all legislation. If you have a legal claim at least make the research and claim something specific.

> Problem is there is NO reason for the user to give etcher any access to the internet at all. If I turn my network connection physically off (turning wifi off and/or unplug my ethernet) etcher will [...]

We never claimed that internet access is required to flash. Of course Etcher works without internet, I'm not sure what argument you're trying to make. The reason we're collecting usage data is to ensure there are no regressions or new bugs in the wild. I've analysed the rationale multiple times in other issues. If a user cares about Etcher there is a legitimate argument to be made that allowing internet access helps a project that you like. And again, we're talking about anonymous usage data here.

Users have undoubtedly benefited from our continued effort to reduce error reports from Etcher. You might not care about other users, that's ok. You might not care that Etcher is as robust as it is, that's also ok. But we care about both of these things which is why the defaults are what they are. You can always change the behaviour in your settings page and Etcher will respect it.

> This is just not acceptable. Not to balena, not to google, not to anyone. All data belongs to me, all data stays on my device. Easy? easy!

This is, like, you know, your opinion. Which for the millionth time, is 100% respected and welcomed. Please disable data collection from your settings. Easy? easy!

petrosagg, 2019-09-20:
> Agreed. Too many developers assume that surveillance is acceptable. “Ethics” needs to be a requirement in computer science courses.

@thefaj totally agreed, ethics absolutely need to be taught. So many people make the mistake of assuming their ethical framework is the only right one and imposing it on others. Very annoying when it happens.

>  There is no acceptable reason to have this malware in the app.

There is no malware in this app. Stop using words in the wrong way to make the issue sound worse. We are collecting anonymous usage data that has unquestionably helped fix all sorts of bugs in Etcher and made it a better app. You might even be benefiting from all this as you use it!

> (And how does somebody “accidentally” include a third party library?)

That's a good question. The way this happened is because etcher loads some content from the internet inside an iframe. This content lives in our marketing website, balena.io, that includes google analytics and other libraries.

Unfortunately, the static site generator we use included these libraries in all the pages, including the ones loaded by etcher, which is how they accidentally got loaded.

We have fixed the inclusion of these now and are preparing for a new release

petrosagg, 2019-09-20:
since the conversation has derailed again about making the data collection opt-in versus the current opt-out situation I'll close this issue in favour of #2766 

For anyone reading the thread from the future, a setting for offline mode already exists in the settings page of Etcher.

closed 2019-09-20 by petrosagg

rradar, 2019-09-29:
@petrosagg wrote:

> For anyone reading the thread from the future, a setting for offline mode already exists in the settings page of Etcher.

...and this "setting" can only be accessed after (**ALL**) **data has been leaked** already... nice "settings" :disappointed:

rradar, 2019-10-15:
And after the existing "offline mode" (@petrosagg) is activated etcher still makes calls to the cloud (which is NOT offline!). Nice setting you have there.. :man_facepalming:

[image]

lurch, 2019-10-15:
@rradar Have you also turned off the update check?
At least it's an https:// connection and not an http:// connection :wink:

rradar, 2019-10-16:
@lurch the only available option regarding updates: "Auto-updates enabled" is **not enabled**:

[image]

The last sentence of you sounds like a bad joke. Kind of: "We are stealing your things but at least do it at night!". Probably every spy- and malware out there uses https...

lurch, 2019-10-16:
Sorry, I didn't mean to cause any offence.

rradar, 2020-03-07:
Found this very [nice little program](https://gitlab.com/bztsrc/usbimager) (250kb) which supports **flashing and verify images privately** without any spyware/adware included. It doesn't transmit any data unlike balena etcher does. Very useful and roughly half a gigabyte saved on my hard disk now that etcher is finally gone :rocket:

rradar, 2020-10-28:
Found this very nice little program called [usbimager](https://gitlab.com/bztsrc/usbimager) (250kb) which supports flashing and verify images privately **without any spyware/adware** included. It doesn't transmit any data unlike balena etcher does. Very useful and roughly half a gigabyte saved on my hard disk now that etcher is finally gone :rocket: