balena-io/etcher #2892: 1.5.56 on Win7, 'STARTING' dies

bbillp, 2019-08-30:
1.5.56 August 21, 2019
Windows 7 
SDR_play_RPI_V0.6.img 
- **Do you see any meaningful error information in the DevTools?** 
No

Etcher executes, I select the input file, Etcher advances to Starting and never shows additional activity but after 10 minutes the FLASH screen reappears.  The SD will open in explorer but is empty.  Before starting Etcher the 32gig Samsung SD was cleared with diskpart then formatted FAT16 in Windows disk manager.

lurch, 2019-08-30:
Took a bit of searching, but I eventually found a link to the mentioned image at https://www.sdrplay.com/community/viewtopic.php?f=6&t=4884 in case it helps the Etcher developers.

bbillp, 2019-09-02:
I found the problem, after starting Etcher and selecting input and output devices and files, START, a Microsoft message pops up behind the Etcher window that cannot be seen, the message needs acknowledgement to execute a MS command then once clicked the SD write continued to completion.     Bill P.

closed 2019-09-02 by thundron