balena-io/etcher #2893: Error during verification 'Something went wrong'

dhammel, 2019-09-02:
- **Etcher version:** 
1.5.56
- **Operating system and architecture:** 
Kubuntu 19.04, x64 bit
- **Image flashed:** 
PopOS 19.04 Intel/Amd amd64
- **Do you see any meaningful error information in the DevTools?** 
The only error shown in dev tools:
/tmp/.mount_EtcherVo…js/dist/raven.js:58 {"stack":"Error: e\n    at Object.exports.createError (/tmp/.mount_EtcherVoigNS/resources/app.asar/lib/shared/errors.js:253:17)\n    at elevateScriptUnix (/tmp/.mount_EtcherVoigNS/resources/app.asar/lib/shared/permissions.js:152:18)","message":"e","image":"pop-os_19.04_amd64_intel_10.iso"}**

This error occurs on every iso write to any usb drive, (2.0, 3.0. Sandisk drives, pny, kingston, different capacity's, different ports and i'm not using any usb hubs.) Even tried different iso images, for kubuntu as well as pop os, even tried an lts version of them. I do always verify my images afer downloading using supplied methods by the developers, they are not corrupted and etcher does successfully write the image to usb as it always has (I can successfully boot from the usbs as usual) , this seems more like a false error but maybe of some concern why i'm getting it. It only happens after the verification stage. It happended after the update to 1.5.56, I forget what version I was running previously, I am using the appimage so usually my appimages are more out of date than software through apt, I usually update all my appimages monthly, if updates are availible.

In case it may be of concern, I am new to the Non-LTS version of Kubuntu, was previously running 18.04 LTS, however etcher on it's previous version worked as usual even on non-lts version of my os until this update to 1.5.56.

lurch, 2019-09-02:
Older versions are available at https://github.com/balena-io/etcher/releases in case you want to see if an older version still works?

dhammel, 2019-09-02:
> Older versions are available at https://github.com/balena-io/etcher/releases in case you want to see if an older version still works?

I was thinking about trying that, however I wasn't sure which version I was previously on. Usually my appimages I update montly, figured it would be a lot of troubl to go through each one of them till I hit one that maybe works, it may/may not be an issue directly with etcher but from what I can see, don't know what else it could be. I can try, at a later time to see if maybe the last 3 or 4 versions work and post back, maybe I'll try the PPA or another package of same version even and see if its an issue just with the appimage, actually just though of that.

jelle619, 2019-09-10:
I am currently facing the exact same issue.

spaceoden, 2019-12-04:
Ubuntu 19.10 here. I had the same issue with the 15.66 AppImage, but the PPA is working for me.

closed 2019-12-08 by dhammel

diegorondini, 2019-12-09:
@dhammel you closed the issue because this has been fixed?

jelle619, 2019-12-09:
I haven’t been noticing this issue lately, but I should also say I’ve switched to a different Linux distribution since then.

dhammel, 2019-12-12:
> @dhammel you closed the issue because this has been fixed?

Yes.

Not sure exactly when it was fixed as i haven't been around too much here lately but I have not been getting this issue since one of the later updates.

SalmonKing72, 2019-12-31:
After updating to macos catalina, I discovered that I had an outdated version of etcher causing this very same issue. I updated to latest (1.5.70 as of this comment) and the problem was resolved.