balena-io/etcher #2894: Popup window says successful completion, main window says mismatched checksum

pdcastro, 2019-09-02:
Creating this issue following feedback from a user in the following balenaEtcher forum thread:
https://forums.balena.io/t/conflicting-messages/29171

I gather that the main issue here is the conflict between the popup that indicates a successful completion, and the main window that reports a mismatched checksum.

I'm inviting the user to fill in the additional details below.

- **Etcher version:** 
- **Operating system and architecture:** 
- **Image flashed:** 
- **Do you see any meaningful error information in the DevTools?**

KEINOS, 2019-09-19:
Same issue on Mac.

- OS: macOS Mojave (OSX 10.14.6)
- Etcher: 1.5.57 (Installed via brew cask)
- Target image used:
    - Raspbian Buster Lite @ https://www.raspberrypi.org/downloads/raspbian/
    - Download SHA-256 checksum matched
- Note: Every time when flashing it asks for the admin password.

I went to the "support thread" link [above](https://github.com/balena-io/etcher/issues/2894#issuecomment-527257068) though, but needs to log-in, so couldn't check.

<details><summary>Message and console log</summary><div><br>

- Main Window

> Flash Complete!
> 1 Failed Device
> /dev/disk3: Source and destination checksums do not match: a1150... !== 80cc4...
> [image]

- Popup

> Flash Comlete!
> 2019-07-10-raspbian-buster-lite.zip was successfully flashed to Generic STORAGE DEVICE Media (/dev/disk3)
> [image]

- Console.log from Developer Tools

```json
/Applications/balenaEtcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:58 Download the React DevTools for a better development experience: https://fb.me/react-devtools
You might need to use a local HTTP server (instead of file://): https://fb.me/react-devtools-faq
/Applications/balenaEtcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:58  _____ _       _
|  ___| |     | |
| |__ | |_ ___| |__   ___ _ __
|  __|| __/ __| '_ \ / _ \ '__|
| |___| || (__| | | |  __/ |
\____/ \__\___|_| |_|\___|_|

Interested in joining the Etcher team?
Drop us a line at join+etcher@balena.io

Version = 1.5.57, Type = dmg
/Applications/balenaEtcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:58 Fri Sep 20 2019 04:05:46 GMT+0900 (JST) Application start ({"packageType":"dmg","version":"1.5.57","applicationSessionUuid":"ba285f02-9e3d-4f96-9fcd-c0b953cd7138","sample":0.1})
/Applications/balenaEtcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:58 Fri Sep 20 2019 04:05:47 GMT+0900 (JST) SafeWebview loaded ({"event":{"fromCache":false,"headers":{"Accept":"text/html,application/xhtml+xml,application/xml;q=0.9,image/webp,image/apng,*/*;q=0.8","Upgrade-Insecure-Requests":"1","User-Agent":"Mozilla/5.0 (Macintosh; Intel Mac OS X 10_14_6) AppleWebKit/537.36 (KHTML, like Gecko) balena-etcher/1.5.57 Chrome/66.0.3359.181 Electron/3.1.9 Safari/537.36"},"id":11,"method":"GET","referrer":"","resourceType":"mainFrame","responseHeaders":{"Accept-Ranges":["bytes"],"Cache-Control":["public, max-age=0"],"Connection":["keep-alive"],"Content-Encoding":["gzip"],"Content-Type":["text/html; charset=UTF-8"],"Date":["Thu, 19 Sep 2019 19:05:46 GMT"],"Etag":["W/\"3c9df-16d40c5ae00\""],"Last-Modified":["Tue, 17 Sep 2019 19:49:20 GMT"],"Server":["Cowboy"],"Transfer-Encoding":["chunked"],"Vary":["Accept-Encoding"],"Via":["1.1 vegur, 1.1 vegur"]},"statusCode":200,"statusLine":"HTTP/1.1 200 OK","timestamp":1568919947349.4958,"url":"https://www.balena.io/etcher/success-banner/?etcher-version=1.5.57&api-version=2&optOutAnalytics=false"},"applicationSessionUuid":"ba285f02-9e3d-4f96-9fcd-c0b953cd7138","flashingWorkflowUuid":"4a59d011-4017-4ab9-af64-ffd0d9fe4862","sample":0.1})
/Applications/balenaEtcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:58 Fri Sep 20 2019 04:12:13 GMT+0900 (JST) Open image selector ({"applicationSessionUuid":"ba285f02-9e3d-4f96-9fcd-c0b953cd7138","flashingWorkflowUuid":"4a59d011-4017-4ab9-af64-ffd0d9fe4862","sample":0.1})
/Applications/balenaEtcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:58 Fri Sep 20 2019 04:12:18 GMT+0900 (JST) Select image ({"image":{"size":2197815296,"compressedSize":426250751,"logo":false,"name":"2019-07-10-raspbian-buster-lite.img","hasMBR":true,"partitions":[{"offset":4194304,"size":268435968,"type":12,"index":1},{"offset":276824064,"size":1920991232,"type":131,"index":2}],"path":"/Users/admin/Downloads/[Image]_Raspbian_Buster/2019-07-10-raspbian-buster-lite.zip","extension":"zip","blockMap":false},"applicationSessionUuid":"ba285f02-9e3d-4f96-9fcd-c0b953cd7138","flashingWorkflowUuid":"4a59d011-4017-4ab9-af64-ffd0d9fe4862","sample":0.1})
/Applications/balenaEtcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:58 Fri Sep 20 2019 04:12:24 GMT+0900 (JST) Open modal ({"name":"drive-selector","applicationSessionUuid":"ba285f02-9e3d-4f96-9fcd-c0b953cd7138","flashingWorkflowUuid":"4a59d011-4017-4ab9-af64-ffd0d9fe4862","sample":0.1})
/Applications/balenaEtcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:58 Fri Sep 20 2019 04:12:29 GMT+0900 (JST) Toggle drive ({"drive":{"size":31914983424,"isVirtual":false,"enumerator":"DiskArbitration","logicalBlockSize":512,"raw":"/dev/rdisk3","error":null,"isReadOnly":false,"displayName":"/dev/disk3","blockSize":512,"isSCSI":false,"isRemovable":true,"device":"/dev/disk3","busVersion":null,"isSystem":false,"busType":"USB","isCard":false,"isUSB":true,"devicePath":"IODeviceTree:/PCI0@0/XHC1@14","mountpoints":[],"description":"Generic STORAGE DEVICE Media","isUAS":null},"previouslySelected":false,"applicationSessionUuid":"ba285f02-9e3d-4f96-9fcd-c0b953cd7138","flashingWorkflowUuid":"4a59d011-4017-4ab9-af64-ffd0d9fe4862","sample":0.1})
/Applications/balenaEtcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:58 Fri Sep 20 2019 04:12:30 GMT+0900 (JST) Modal accepted ({"name":"drive-selector","applicationSessionUuid":"ba285f02-9e3d-4f96-9fcd-c0b953cd7138","flashingWorkflowUuid":"4a59d011-4017-4ab9-af64-ffd0d9fe4862","sample":0.1})
[... cut ...]
/Applications/balenaEtcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:58 Fri Sep 20 2019 04:20:38 GMT+0900 (JST) Navigate ({"to":"success","from":"main","applicationSessionUuid":"ba285f02-9e3d-4f96-9fcd-c0b953cd7138","sample":0.1})
/Applications/balenaEtcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:58 [withTheme] You are not using a ThemeProvider nor passing a theme prop or a theme in defaultProps in component class "Component"
console.(anonymous function) @ /Applications/balenaEtcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:58
```

</div></details>

KEINOS, 2019-09-20:
It seems that all 3 of the brand new SanDisk **microSD cards that I bought were a defective product** in the first place... 

I validated the SD card with the `f3` command on Mac, which is somewhat similar to `H2testw` on Windows, and noticed that the data gets corrupted on reading.

In addition, the 4th one, which I just bought worked so fine ... (Transcend SD card 32GB UHS-I Class10 TS32GSDC300S-E)

So, this **"Mismatch checksum" issue seems to happen when no error happens while writing but reading, and has a bad sector** or something.

<details><summary>Log of validating SD Card on macOS</summary><div><br>

```shellsession
$ # Env info
$ sw_vers
ProductName:	Mac OS X
ProductVersion:	10.14.6
BuildVersion:	18G95
$ brew install f3
...
$ f3write -V
F3 Write 7.2
$ f3read -V
F3 Read 7.2
$
$ # Find SD Card Writer's disk number
$ diskutil list
...
$ # Format SD Card (disk3) with the disk name "RPI"
$ diskutil eraseDisk MS-DOS RPI disk3
Started erase on disk3
Unmounting disk
Creating the partition map
Waiting for partitions to activate
Formatting disk3s2 as MS-DOS (FAT) with name RPI
512 bytes per physical sector
/dev/rdisk3s2: 61891008 sectors in 1934094 FAT32 clusters (16384 bytes/cluster)
bps=512 spc=32 res=32 nft=2 mid=0xf8 spt=32 hds=255 hid=411648 drv=0x80 bsec=61921280 bspf=15111 rdcl=2 infs=1 bkbs=6
Mounting disk
Finished erase on disk3
$
$ # Validation (Writing)
$ time f3write /Volumes/RPI
F3 write 7.2
Copyright (C) 2010 Digirati Internet LTDA.
This is free software; see the source for copying conditions.

Free space: 29.51 GB
Creating file 1.h2w ... OK!                         
Creating file 2.h2w ... OK!                         
Creating file 3.h2w ... OK!                          
Creating file 4.h2w ... OK!                          
Creating file 5.h2w ... OK!                          
Creating file 6.h2w ... OK!                          
Creating file 7.h2w ... OK!                          
Creating file 8.h2w ... OK!                          
Creating file 9.h2w ... OK!                          
Creating file 10.h2w ... OK!                          
Creating file 11.h2w ... OK!                          
Creating file 12.h2w ... OK!                          
Creating file 13.h2w ... OK!                          
Creating file 14.h2w ... OK!                          
Creating file 15.h2w ... OK!                          
Creating file 16.h2w ... OK!                          
Creating file 17.h2w ... OK!                          
Creating file 18.h2w ... OK!                          
Creating file 19.h2w ... OK!                          
Creating file 20.h2w ... OK!                          
Creating file 21.h2w ... OK!                          
Creating file 22.h2w ... OK!                          
Creating file 23.h2w ... OK!                         
Creating file 24.h2w ... OK!                         
Creating file 25.h2w ... OK!                         
Creating file 26.h2w ... OK!                         
Creating file 27.h2w ... OK!                         
Creating file 28.h2w ... OK!                         
Creating file 29.h2w ... OK!                         
Creating file 30.h2w ... OK!                        
Free space: 0.00 Byte
Average writing speed: 13.45 MB/s

real	37m27.986s
user	0m7.668s
sys	0m41.088s
$
$ # Validating (Reading)
$ time f3read /Volumes/RPI
F3 read 7.2
Copyright (C) 2010 Digirati Internet LTDA.
This is free software; see the source for copying conditions.

                  SECTORS      ok/corrupted/changed/overwritten
Validating file 1.h2w ... 1920363/    45393/      0/ 131396
Validating file 2.h2w ... 2097152/        0/      0/      0
Validating file 3.h2w ... 2097152/        0/      0/      0
Validating file 4.h2w ... 2097152/        0/      0/      0
Validating file 5.h2w ... 1948216/    58312/      0/  90624
Validating file 6.h2w ... 2097152/        0/      0/      0
Validating file 7.h2w ... 2097152/        0/      0/      0
Validating file 8.h2w ... 2097152/        0/      0/      0
Validating file 9.h2w ... 1883045/   103316/      0/ 110791
Validating file 10.h2w ... 2097152/        0/      0/      0
Validating file 11.h2w ... 2097152/        0/      0/      0
Validating file 12.h2w ... 2097152/        0/      0/      0
Validating file 13.h2w ... 1931366/    62452/      0/ 103334
Validating file 14.h2w ... 2097152/        0/      0/      0
Validating file 15.h2w ... 2097152/        0/      0/      0
Validating file 16.h2w ... 2097152/        0/      0/      0
Validating file 17.h2w ... 1956032/    66618/      0/  74502
Validating file 18.h2w ... 2097152/        0/      0/      0
Validating file 19.h2w ... 2097152/        0/      0/      0
Validating file 20.h2w ... 2097152/        0/      0/      0
Validating file 21.h2w ... 1980766/    63769/      0/  52617
Validating file 22.h2w ... 2097152/        0/      0/      0
Validating file 23.h2w ... 2097152/        0/      0/      0
Validating file 24.h2w ... 2097152/        0/      0/      0
Validating file 25.h2w ... 1959648/    61712/      0/  75792
Validating file 26.h2w ... 2097152/        0/      0/      0
Validating file 27.h2w ... 2097152/        0/      0/      0
Validating file 28.h2w ... 2097152/        0/      0/      0
Validating file 29.h2w ... 1974024/    46287/      0/  76841
Validating file 30.h2w ... 1070432/        0/      0/      0

  Data OK: 28.93 GB (60664084 sectors)
Data LOST: 597.54 MB (1223756 sectors)
	       Corrupted: 247.98 MB (507859 sectors)
	Slightly changed: 0.00 Byte (0 sectors)
	     Overwritten: 349.56 MB (715897 sectors)
Average reading speed: 62.86 MB/s

real	8m0.749s
user	0m17.038s
sys	0m26.167s

```

</div></details>

thundron, 2019-09-25:
@KEINOS Thank you for taking the time to investigate, unfortunately it's not rare to see a corrupted sd card, it sounds strange though that it happened on 3 brand-new sd cards. Did you try with a different card reader too to lure that out?

lurch, 2019-09-25:
Unless they were all 'fake' cards, and he bought them from the same vendor at the same time? :man_shrugging: 
If they're not fake, and f3 reports that they're defective, you should be able to replace them under warranty?

KEINOS, 2019-09-26:
@thundron @lurch  Indeed. Not surprised about corrupted SD cards, they were same brand and same lot. Since we had a shot to record highs during the summer heat in Japan, I think that caused the corruption to the cards in the warehouse. I wasn't lucky enough. 🤷‍♂

But the point is, this "checksum error says success on popup" issue seems to occur when:

1. Card writing was a success.
2. Card reading (checksum compare) was a failure.

So, the message itself is not lying, since the writing (flash) process was a success.
But if the popup message says something like below, the user might notice it better, I think.

> Flash Complete! (with errors)
> xxxxx.zip was successfully flashed to Generic STORAGE DEVICE Media (/dev/diskX) 
> But 1 error detected.

KEINOS, 2019-09-26:
@lurch I bought them via Amazon and I'm asking the supplier about the warranty but no response so far. 😭

lurch, 2019-09-26:
> I bought them via Amazon and I'm asking the supplier about the warranty but no response so far.

OffTopic, but I think if you don't get a response from the supplier (within a certain number of days), you should be able to get a refund from Amazon?

saulpalv, 2020-05-28:
I also have 2 sandisk cruzer blade 16 gb usb and both show checksum error for ubuntu iso
[image]

thundron, 2020-05-29:
@saulpalv as mentioned, it's probably a corrupt media. Please try with another sd card/usb stick and let us know how it goes, if there are any other issues open a new one. This issue is about the discrepancy of the alert vs the final screen

Spokehedz, 2020-06-04:
This is really unclear, and very confusing to people who don't know much about computers--which are the people that Etcher is targeted towards.

[image]