balena-io/etcher #2900: RPM installation fails on Fedora Silverblue

zamber, 2019-09-09:
- **Etcher version:** Latest (v1.5.56)
- **Operating system and architecture:** Fedora Silverblue 30

The RPM does not install properly.

```➜ rpm-ostree install balena-etcher-electron              
Checking out tree c1d33ee... done
Enabled rpm-md repositories: updates fedora-cisco-openh264 fedora etcher-rpm
rpm-md repo 'updates' (cached); generated: 2019-09-09T01:56:55Z
rpm-md repo 'fedora-cisco-openh264' (cached); generated: 2019-03-21T15:16:16Z
rpm-md repo 'fedora' (cached); generated: 2019-04-25T23:49:41Z
rpm-md repo 'etcher-rpm' (cached); generated: 2019-07-22T10:35:47Z
Importing rpm-md... done
Resolving dependencies... done
Checking out packages... done
Running pre scripts... done
Running post scripts... done
error: Running %post for balena-etcher-electron: Executing bwrap(/bin/sh): Child process killed by signal 1; run `journalctl -t 'rpm-ostree(balena-etcher-electron.post)'` for more information
```

```➜ journalctl -t 'rpm-ostree(balena-etcher-electron.post)'
-- Logs begin at Fri 2019-08-23 07:18:27 CEST, end at Mon 2019-09-09 12:24:43 CEST. --
Sep 06 10:57:00 synergy rpm-ostree(balena-etcher-electron.post)[2404]: ln: failed to create symbolic link '/usr/local/bin/balena-etcher-electron': No such file or directory
-- Reboot --
Sep 09 12:00:55 synergy rpm-ostree(balena-etcher-electron.post)[2445]: ln: failed to create symbolic link '/usr/local/bin/balena-etcher-electron': No such file or directory
Sep 09 12:02:43 synergy rpm-ostree(balena-etcher-electron.post)[2445]: ln: failed to create symbolic link '/usr/local/bin/balena-etcher-electron': No such file or directory

lurch, 2019-09-16:
AFAIK `.rpm` files shouldn't be trying to create things in `/usr/local/`.
There was a similar problem with `.deb` files in #2458

ccarguelo-uca, 2019-11-04:
> * **Etcher version:** Latest (v1.5.56)
> * **Operating system and architecture:** Fedora Silverblue 30
> 
> The RPM does not install properly.
> 
> ```
> Checking out tree c1d33ee... done
> Enabled rpm-md repositories: updates fedora-cisco-openh264 fedora etcher-rpm
> rpm-md repo 'updates' (cached); generated: 2019-09-09T01:56:55Z
> rpm-md repo 'fedora-cisco-openh264' (cached); generated: 2019-03-21T15:16:16Z
> rpm-md repo 'fedora' (cached); generated: 2019-04-25T23:49:41Z
> rpm-md repo 'etcher-rpm' (cached); generated: 2019-07-22T10:35:47Z
> Importing rpm-md... done
> Resolving dependencies... done
> Checking out packages... done
> Running pre scripts... done
> Running post scripts... done
> error: Running %post for balena-etcher-electron: Executing bwrap(/bin/sh): Child process killed by signal 1; run `journalctl -t 'rpm-ostree(balena-etcher-electron.post)'` for more information
> ```
> 
> ```
> -- Logs begin at Fri 2019-08-23 07:18:27 CEST, end at Mon 2019-09-09 12:24:43 CEST. --
> Sep 06 10:57:00 synergy rpm-ostree(balena-etcher-electron.post)[2404]: ln: failed to create symbolic link '/usr/local/bin/balena-etcher-electron': No such file or directory
> -- Reboot --
> Sep 09 12:00:55 synergy rpm-ostree(balena-etcher-electron.post)[2445]: ln: failed to create symbolic link '/usr/local/bin/balena-etcher-electron': No such file or directory
> Sep 09 12:02:43 synergy rpm-ostree(balena-etcher-electron.post)[2445]: ln: failed to create symbolic link '/usr/local/bin/balena-etcher-electron': No such file or directory
> ```

@zamber Fedora Silverblue is not meant for installing .rpm packages, kindly check Fedora Silverblue documentation for that. [Fedora-Silverblue/Getting-Started](https://docs.fedoraproject.org/en-US/fedora-silverblue/getting-started/)

> **Flatpak is the primary way that apps can be installed on Silverblue. (For information, see flatpak.org.) Flatpak works out of the box in Fedora Silverblue, and Fedora provides a small (but growing) collection of apps that can be installed.**

> **The other main source of Flatpak apps is Flathub, which provides a large repository of Flatpak apps that can be installed.**

> **Package layering works by modifying your Silverblue installation. As the name implies, it works by extending the packages from which Silverblue is composed. Most (but not all) RPM packages provided by Fedora can be installed on Silverblue using this method.**

zamber, 2019-11-04:
Yep it's not recommended but possible for such cases. Balena Etcher is not
available in any other form, so installing it from an RPM is the only way
to install it on Silverblue. It's a fallback way to get stuff working that
would otherwise not work, like RPM Fusion repos.

This also does not justify spawning shell commands after installation that
modify system directories. I have my doubts if uninstalling on regular
Fedora actually cleans up properly due to this.

This is generally bad packaging practice, not an issue with Fedora
(Silverblue or otherwise) itself. Silverblue is just more strict about it.

lurch, 2019-11-04:
> Flatpak is the primary way that apps can be installed on Silverblue.

See #2019 for more info on that.

Stupendoussteve, 2020-01-19:
> Balena Etcher is not available in any other form, so installing it from an RPM is the only way to install it on Silverblue.

The Etcher AppImage works fine on SilverBlue.

closed 2023-07-12 by dfunckt