balena-io/etcher #2902: Mixpanel, Google Analytics, etc spyware!!!

thefaj, 2019-09-14:
- **Etcher version:** 1.5.x
- **Operating system and architecture:** Mac
- **Image flashed:** Yup
- **Do you see any meaningful error information in the DevTools?** Nope

Why is this app sending my private data to Mixpanel, Google Analytics, Google Tag Manager, and resin.io without my knowledge???

RESPECT USER PRIVACY and tell us why you want to send our personally identifiable data to these surveillance companies. Then give us the ability to *OPT IN*.

tuxayo, 2019-09-16:
Can you detail how did you find the list of third parties involved?

tuxayo, 2019-09-16:
Issues related:
- #2903
- #2599
- #2766

petrosagg, 2019-09-16:
@thefaj  see the detailed response here https://github.com/balena-io/etcher/issues/2766#issuecomment-531437542 but the tl;dr is that we intend to collect **anonymous** usage data for the improvement of etcher. Any personally identifiable data sent is a bug and we're currently investigating with our legal team which data is PII and which isn't.

Closing as duplicate of #2766, let's keep the discussion in one place

closed 2019-09-16 by petrosagg

thefaj, 2019-09-16:
Yeah…I fully disagree with that comment. I'd say you should keep this issue open since it makes it more apparent to users that you're giving their personal information over to many large surveillance companies without their knowledge. There's nothing **anonymous** about including Google Analytics or Mixpanel. It might just appear anonymous to whoever has access to the associated control panels. Anything sent over is personally identifiable data…it's a bummer that you've chosen to sell out users of this tool for a vanity statistic.

In the meantime, the macOS terminal command "dd" is safe to use. More info here for those who have seen this issue too: https://www.raspberrypi.org/documentation/installation/installing-images/mac.md

petrosagg, 2019-09-16:
> There's nothing anonymous about including Google Analytics or Mixpanel.

As detailed in #2766, Google Analytics was included unintentionally and is being removed. 

> Anything sent over is personally identifiable data

No, this couldn't be more wrong. Personally Identifiable Data is data that you can give to someone and tell you "this looks like thefaj". It's not a subjective opinion you have over the data. If Etcher loads mixpanel and starts sending random numbers as events, it's not personal. If Etcher loads mixpanel and sends the time it took to flash a card of a particular size, it's not personal. If Etcher loads mixpanel and sends the username of your computer, it's personal. 

As also detailed in #2766, we're not interested in personally identifiable data. We don't care what images **you** write or when you wrote them. What we care about is "Did this new release of etcher perform as well or better than the previous version?".

> I'd say you should keep this issue open since it makes it more apparent to users that you're giving their personal information over to many large surveillance companies without their knowledge.

I'm sorry, but you're misrepresenting reality. We're only using Sentry (for when Etcher crashes) and Mixpanel (to track things like flashing speed, success to failure ratios, etc). We're neither sending personal data nor selling to any nefarious ad network to sell you stuff.

thefaj, 2019-09-16:
I'm well aware of what Mixpanel and Google Analytics are capable of. An IP address is personal information. It's really awful that you're defending the inclusion of this spyware instead of admitting this was a poor decision and choosing to remove it.

Don't send any network data without allowing your users to opt-in. This is about basic user respect—privacy and security are not about avoiding ad networks.

petrosagg, 2019-09-16:
> I'm well aware of what Mixpanel and Google Analytics are capable of.

As I mentioned above, we're not using Google Analytics

> An IP address is personal information.

That's true, which is why we don't log the IP addresses https://help.mixpanel.com/hc/en-us/articles/360000679006-Managing-Personal-Information#disable-geolocation-tracking

> This is about basic user respect—privacy and security are not about avoiding ad networks.

There are many aspects of caring for the users. Having a robust piece of software that cares about fixing obscure edge cases in one of them. Respecting their privacy and security considerations is another one.

**Etcher currently does both.** You can disable data collection from your settings page if that's what you want. This option exists and will continue to exist.

If you have a constructive comment about how to balance performance and error report benefits with  making the data collection opt-in benefits please join the discussion on #2766 

Shouting your moral positions on what is the right thing and hugely misrepresenting reality (calling Etcher spyware is absurd in so many levels) won't get very far. You have to acknowledge that both options are damaging users in some way. It maybe be a clear choice for YOU, but we care about the users as a whole, not who shouts the loudest.

thefaj, 2019-09-16:
It's good that you've made your position public here for others to read. This software is not safe to use, and you have no intention to fix it. Best of luck.

petrosagg, 2019-09-16:
> you have no intention to fix it.

I fail to see how an open issue about discussing making it opt-in is "no intention to fix it" but I can't help with that :)

petrosagg, 2019-09-16:
locking in favour of #2766