balena-io/etcher #2903: Remove or create opt-in setting for advertisements

nyanshak, 2019-09-14:
- **Etcher version:** 1.4.9+
- **Operating system and architecture:** any
- **Image flashed:** any
- **Do you see any meaningful error information in the DevTools?** no

I'd like the team to consider removing ads from teh tool. If not, consider making ads opt-in via the settings panel. I find the advertisements distracting and annoying when monitoring image flash progress.

Partial workaround to remove some of the advertisements:
* In ${XDG_CONFIG_HOME}/balena-etcher-electron/config.json, add this line:
featuredProjectEndpoint": "http://127.0.0.1"

The discussion began in issue #2599, but got closed referencing #2766, which I find to be a completely separate issue about data tracking.

> Since the original question about removing the feature projects was addressed and the discussion moved to data collection concerns I'm closing this in favour of #2766

It was discussed, but the team did not make any announcements / decisions about this, leaving it an open question / feature request.

petrosagg, 2019-09-14:
> It was discussed, but the team did not make any announcements / decisions about this, leaving it an open question / feature request.

Indeed the previous thread was a bit messy and I should have been a bit clearer on what the resolution was.

As my co-founder Alex commented [here](https://github.com/balena-io/etcher/issues/2599#issuecomment-456225518), we consider the featured project page to be meaningful and interesting to the people that use etcher while at the same time helping promote our products that help fund the development of etcher. What we include there is high quality content carefully built by our team that guides you through building a particular project. The gap between this and irrelevant ads from an ad network is vast.

For the above reasons we are not interested in removing it or making it opt-in for the time being.

closed 2019-09-14 by petrosagg

Kurgol, 2019-09-16:
I have just replaced etcher on my system now - it's obvious the devs don't give a crap what the users want. 

Startup Disk Creator is working perfectly, with no spammy sponsorship adverts.

rradar, 2019-09-16:
I also don't use etcher anymore before this is fixed.

Any idea @Kurgol if the Startup Disk Creator is validating the images flashed?

EpicLPer, 2019-09-29:
I've opened yet another issue regarding this cause they closed and locked the old thread. New one is #2914 .