balena-io/etcher #2904: Operation not permitted.

bl1mp, 2019-09-14:
- v1.5.56
- 10.15 Beta Catalynia
- ubuntu-19.04-desktop-amd64.iso to usb formatted as MSDOS scheme: GUID partition map
- **Do you see any meaningful error information in the DevTools?** 
No, it runs as expected, until flash, then it isn't permitted it says, changing permissions, giving etcher full access, changing permissions on disk, logging to root, nothing works.
"Process has ended unexpectedly" 

[image]

lurch, 2019-09-16:
https://github.com/balena-io/etcher/search?q=catalina&type=Issues might have some workarounds?

bl1mp, 2019-09-19:
### I have managed to flash the usb with ubuntu

This seems to be an issue in the catalynia beta, it seems. 

I have tried to repeat these steps running Mojave, no settings had to be adjusted, it worked perfectly fine. I'm not sure why it won't give permission, even if you change it, but apple really tightened it up too much in cata.

closed 2019-09-19 by bl1mp

lurch, 2019-09-19:
One of the Etcher developers recently said they've got plans in place
https://github.com/balena-io/etcher/issues/2887#issuecomment-532716491

leadersm, 2019-12-16:
i solve the problem with version : 1.5.69 
https://www.macdown.com/mac/5303.html

lurch, 2019-12-16:
@leadersm Yes, it was fixed by #2939