balena-io/etcher #2905: Armbian *.7z support

iav, 2019-09-16:
Will be good if Etcher can directly open armbian *.7z images.

lurch, 2019-09-16:
Duplicate of #711 ?

iav, 2019-09-17:
yes — if look to first message. But then you go deep into complicated internal image properties, linux kernel compression and so on.
I now ask about simply decompress image before writing, no relation to kernels, kernel compression algorhytm support etc.
Just possibility select  .7z file  as "source image" file.

thundron, 2019-09-17:
As mentioned in the issue #711 though,

> We've looked into it a while back, but we couldn't find any decent 7zip
NodeJS modules we could use (all of them either depend on the 7zip tool
being available on the path, or embed the binary inside the package).

And furthermore

> Lets close this for now, as it turns out to be very difficult to find 7-zip implementations we can use.

Closing as duplicate (for now)

closed 2019-09-17 by thundron