balena-io/etcher #2907: "contact the Etcher team if the problem persists"

404ClickBait, 2019-09-17:
- **Etcher version:** 1.5.57
- **Operating system and architecture:** Windows 10
- **Image flashed:** hassbian version 2019-07-02
- **Do you see any meaningful error information in the DevTools?** 

"The writer process ended unexpectedly.  Please try again, and contact the Etcher team if the problem persists"

It's fine, I'll just continue using Win32DiskImager.  Much more stable...

lurch, 2019-09-17:
See also: https://github.com/balena-io/etcher/search?q=The+writer+process+ended+unexpectedly&type=Issues

thundron, 2019-09-18:
@404ClickBait If you're willing to paste here the logs from the application, we could try to fix/reproduce the issue, otherwise it's understandable and we can close this I guess in favor of already open issues

404ClickBait, 2019-09-18:
Let me know if you would like me to get this for your benefit.  To be honest I already have a tool (or two) that do everything I need so it would just be to help you out.

Best regards,
Kyle

lurch, 2019-09-18:
@404ClickBait You might want to edit your email address out of that last message, to stop it being picked up by spambots?

404ClickBait, 2019-09-19:
Hello,

I’m sorry, what does that mean – to “edit your email address out of that last message” – is this email posted publicly?

Thanks,
kyle

lurch, 2019-09-19:
Yes, see https://github.com/balena-io/etcher/issues/2907

404ClickBait, 2019-09-19:
> Yes, see #2907

Thank you for letting me know!

Xu-TikT, 2020-08-12:
1.5.102 Mac version works for me

thundron, 2020-08-12:
Thanks for chiming in @Xu-TikT, although this issue is about a different OS (Windows 10).
It's clear though that there have been several other versions released since this issue was opened, so I'll close this.

FWIW usually (not always of course) that error comes from some specific settings on Windows that prevent Etcher from working.

closed 2020-08-12 by thundron