balena-io/etcher #2908: How does etcher do it's "validated flashing"?

DefunctLizard, 2019-09-21:
I have been interested in etcher recently, and I was wondering what validated flashing is. On the website it says that it checks that all the data is the same on the file and on the disk, but where in the code does it actually do this? Is it a function in DD? or is it something written in the many languages that make up etcher?

lurch, 2019-09-21:
Etcher doesn't use `dd`, the whole thing is written in Javascript.
Since I last worked on Etcher it's gone through quite a big refactoring, and I _think_ the validation code is currently somewhere in https://github.com/balena-io-modules/etcher-sdk

Etcher team: Feel free to delete this comment if it's inaccurate!

thundron, 2019-09-23:
@lurch Correct, the writing/verifying logic has been moved in the etcher-sdk and the code you're looking for is probably [this one](https://github.com/balena-io-modules/etcher-sdk/blob/819182a747acb8372f6ee7353c1ea1da976f0eec/lib/sparse-stream/shared.ts#L112-L126): that's where the verification is actually made by comparing the `checksum`s. That is called by the etcher-sdk writing methods whenever validation is enabled.
Closing, but feel free to ask for more clarifications if needed!

closed 2019-09-23 by thundron

lurch, 2019-09-23:
@thundron I'm not very familiar with TypeScript, but would it make sense for that function to raise an error if both `generateChecksums` and `verify` are true? :man_shrugging:

thundron, 2019-09-23:
@lurch Sure, it's just done a level above [before the actual function call](https://github.com/balena-io-modules/etcher-sdk/blob/819182a747acb8372f6ee7353c1ea1da976f0eec/lib/sparse-stream/shared.ts#L91-L93