balena-io/etcher #2909: EACCES: permission denied

AdamKearn, 2019-09-24:
- **Etcher version:** TOT
- **Operating system and architecture:** Ubuntu 19.04

Hey, I am trying to compile my own version for my own project.
When running the `make electron-develop` command I get the following error.

What do you think is happening?

Hear are my node/npm versions if any of that helps.
```
adam@ubuntu:~/etcher$ npm -v
6.11.3
adam@ubuntu:~/etcher$ node -v
v10.15.2
```

```
Installing dependencies
npx: installed 426 in 8.214s

> electron-chromedriver@3.0.0 install /home/adam/etcher/node_modules/electron-chromedriver
> node ./download-chromedriver.js

/home/adam/etcher/node_modules/electron-chromedriver/download-chromedriver.js:21
    if (error != null) throw error
                       ^

Error: EACCES: permission denied, mkdir '/home/adam/etcher/node_modules/electron-chromedriver/bin'
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! electron-chromedriver@3.0.0 install: `node ./download-chromedriver.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the electron-chromedriver@3.0.0 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/adam/.npm/_logs/2019-09-24T15_09_00_964Z-debug.log
make: *** [Makefile:97: electron-develop] Error 1
adam@ubuntu:~/etcher$ sudo make electron-develop
```

AdamKearn, 2019-09-24:
@lurch  thanks for the super quick response.
If I don't run it as `sudo` I get this in the output.

`
┌──────────────────────────────────────┐
│                               npm update check failed                            │
│                     Try running with sudo or get access                     │
│                     to the local update config store via                      │
│ sudo chown -R $USER:$(id -gn $USER) /home/adam/.config  │
└──────────────────────────────────────┘
`

lurch, 2019-09-24:
I'm only an Etcher-helper, not an Etcher-developer, and I'm afraid this is now beyond my level of knowledge :wink:
Hopefully one of the Etcher developers will reply to this soon? (although I wonder if this is an npm problem rather than an Etcher problem? - did you try running the suggested `sudo chown ...` command?)

thundron, 2019-09-25:
@AdamKearn You shouldn't need to run `make electron-develop` with `sudo`, did you try giving a go at the suggestion from `npm` itself `sudo chown ..` as @lurch mentioned?

AdamKearn, 2019-10-21:
@lurch @thundron I am sooo sorry for the late reply,  I have been super busy with work etc.

I have got Etcher working now,  I was just being stupid.

Anyway, I don't supose you could show me how to export my code to an `.exe` file now?

lurch, 2019-10-21:
Read https://github.com/balena-io/etcher/blob/master/docs/CONTRIBUTING.md and then have a look at https://github.com/balena-io/etcher/blob/master/Makefile and https://github.com/balena-io/scripts/tree/master/electron ? :man_shrugging:

closed 2023-07-12 by dfunckt