balena-io/etcher #2910: Slow download

TheEpicNoobZilla, 2019-09-24:
Hi i have issue with download speed (20-50 KB/s), the problem is not with me bcs steam, uplay or other programs downloaded through browser goes with about 1/2 MB/s

lurch, 2019-09-24:
See #2083 and #2336

TheEpicNoobZilla, 2019-09-25:
Maybe it was my internet, because today it's downloading with 300/400 KB/s

closed 2019-09-25 by TheEpicNoobZilla

myhd, 2020-01-02:
Super slow 1990s modem speed with Github download from Germany. Estimated download time of 14 hours. VPN switch to Netherlands fixed it. But why?

gbirke, 2020-12-19:
@myhd If you are a customer of Deutsche Telekom, it's very likely that it's because they have monopolistic peering agreements with the rest many ISPs (they want everyone to pay them for traffic, instead of saying "I'll route yours if you route mine, in the end it'll balance out"). Their connection the the big connection Hub DE-CIX is 1/20th of what other providers have. Your VPN provider probably has both a favorable peering agreement and a good connection to DE-CIX