balena-io/etcher #2911: “balenaEtcher” can’t be opened because Apple cannot check it for malicious software.

e3b0c442, 2019-09-25:
- **Etcher version:** 1.5.57
- **Operating system and architecture:** macOS Catalina 10.15 beta build 19A573a
- **Image flashed:** n/a
- **Do you see any meaningful error information in the DevTools?** n/a

Etcher will not open on the latest macOS beta, with the error:
“balenaEtcher” can’t be opened because Apple cannot check it for malicious software.

This was downloaded fresh tonight.

Not sure what else I can provide for troubleshooting as macOS prevented the application from even launching.

lurch, 2019-09-25:
See https://github.com/balena-io/etcher/issues/2833#issuecomment-520375994

thundron, 2019-09-25:
Closing in favor of the other issue open, also see https://github.com/balena-io/etcher/issues/2833#issuecomment-511152047 for a temp workaround

e3b0c442, 2019-09-25:
I feel like this may be a different issue. The symptom I am describing is not laid out in any of the comments there, and that workaround does not work; `sudo` has the same result.

thundron, 2019-09-25:
Hmmm, right @e3b0c442 . A quick search led me to this https://apple.stackexchange.com/questions/366542/install-spotify-cant-be-opened-because-apple-cannot-check-it-for-malicious-so so it looks like we'll have to wait a bit for that (app notarisation) to happen on Etcher. I'll keep this open then until we update that part of the OSX package

michaelfm1211, 2019-10-09:
If you right click(Control Click) on the Etcher app then click "Open" you will get a different dialog which has the option to continue to open Etcher.

dtaivpp, 2019-10-21:
Hey all, here is a workaround to install Etcher:

`xattr -d com.apple.quarantine Downloads/<your installer>.pkg`

After executing this line which is pointed at your installer image you will be able to run it. It deletes the 'quarentine' attribute that Apple has set that is disallowing it from running.

thundron, 2019-10-21:
@dtaivpp Correct, the issue will stay open until #2939 (or a substitute of that w/ Electron 6 that we're working on as well) will be ready/merged

gorsim, 2019-10-25:
> Hey all, here is a workaround to install Etcher:
> 
> `xattr -d com.apple.quarantine Downloads/<your installer>.pkg`
> 
> After executing this line which is pointed at your installer image you will be able to run it. It deletes the 'quarentine' attribute that Apple has set that is disallowing it from running.
> 
> @thundron I think you should keep this one open as #2833 is a disk permissions error and this one is related to the app being registered with Apple.

Thanks for this.  Do you put the full name of the .dmg file in the <your installer>?  eg `xattr -d com.apple.quarantine Downloads/<balenaEtcher-1.5.59.dmg>.pkg`

dtaivpp, 2019-10-25:
@gorsim for the dmg you would remove.pkg and just use .dmg extension

gorsim, 2019-10-25:
> @gorsim for the dmg you would remove.pkg and just use .dmg extension

Thanks David.  Annoyingly it didn't work - still said the same thing.  This is the terminal dialogue - the first time it did seem to do something then when I tried again said no such xattr.  I ran it then re-installed balenaEtcher-1.5.59.dmg and when I opened it the message said “balenaEtcher” can’t be opened because Apple cannot check it for malicious software.

 Any ideas?  (Sorry I am not very used to command mode!)

simons-MBP:~ simonnonadmin$ xattr -d com.apple.quarantine Downloads/balenaEtcher-1.5.59.dmg
Simons-MBP:~ simonnonadmin$ xattr -d com.apple.quarantine Downloads/balenaEtcher-1.5.59.dmg
xattr: Downloads/balenaEtcher-1.5.59.dmg: No such xattr: com.apple.quarantine

dtaivpp, 2019-10-25:
@gorsim the first time it worked and removed the attribute. The second time you ran it, it was saying there is no such attribute because it had already been deleted. 

I would download, run the command once, and then attempt to open the dmg. I am unable to replicate any more as it seems my computer recognizes it as safe now

gorsim, 2019-10-25:
> @gorsim the first time it worked and removed the attribute. The second time you ran it, it was saying there is no such attribute because it had already been deleted.
>
> I would download, run the [...]

Thank you so much David - it worked!

gorsim, 2019-10-25:
David - next problem I'm trying to convert CentOS-Userland-7-armv7hl-RaspberryPI-Minimal-4-1908-sda.raw.xz into an image file for a raspberry pi 4 (from herehttp://mirrors.coreix.net/centos-altarch/7.7.1908/isos/armhfp/)  but Etcher says it is corrupted.  I tried another file and it says the same thing.  Does yours work properly?

whatwouldleedo, 2019-10-25:
> * **Etcher version:** 1.5.57
>
>     * **Operating system and architecture:** macOS Catalina 10.15 beta build 19A573a
>
>     * **Image flashed:** n/a
>
>     * **Do you see any meaningful error information [...]

Did you visit your System Preferences > Security & Privacy > then click allow. Try this method after you try to open your file.

gorsim, 2019-10-25:
> Did you visit your System Preferences > Security & Privacy > then click allow. Try this method after you try to open your file.

Thanks for your help - actually I sussed it out via another forum.  You need to run etcher using sudo when logged in as administrator then it works.  Cheers

zvin, 2019-11-06:
Duplicate of #2833

closed 2019-11-06 by zvin