balena-io/etcher #2912: My SD card doesnt work

MegaBloopy, 2019-09-26:
- Etcher version: Newest
- Operating system and architecture: Windows 10 64 Bit
- Image flashed: Linux
- Do you see any meaningful error information in the DevTools? Can't Access 

- So i cant access my 128GB SD card anymore, it says please insert something into drive. and when i plug it in it says directory is invalid

lurch, 2019-09-26:
https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#recovering-broken-drives

MegaBloopy, 2019-09-26:
That doesn’t work, diskpart doesn’t load unless I unplug the Ed card

MegaBloopy, 2019-09-26:
I’ve tried literally everything I can

lurch, 2019-09-26:
Try a different card-reader and/or different computer and/or different OS? :man_shrugging:

MegaBloopy, 2019-09-26:
Havent tried linux but mac and windows and switch dont work

MegaBloopy, 2019-09-26:
Mac worked with some jiggling and modifying i could get an empty 126.6 GB SD card thanks

closed 2019-09-26 by MegaBloopy