balena-io/etcher #2913: Etcher not work on macOS Catalina

mrdong916, 2019-09-29:
- **Etcher version: 1.5.57** 
- **Operating system and architecture: macOS** 
- **Image flashed: https://mirrors.tuna.tsinghua.edu.cn/deepin-cd/15.11/deepin-15.11-amd64.iso** 
- **Do you see any meaningful error information in the DevTools?** 

Error: EPERM: operation not permitted, open '/dev/rdisk2

lurch, 2019-10-02:
Duplicate of #2833 ?

alexmuntean, 2019-10-09:
i managed to flash by copying inside balenaetcher app itself and selecting it from there and disabling system integrity protection (command+r while restarting -> terminal -> csrutil disable)

mrdong916, 2019-10-10:
> Duplicate of #2833 ?

Different versions

MarcGodard, 2019-10-11:
Just tried the latest version and not working.

satmandu, 2019-10-11:
FYI Running ```sudo /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher``` appears to work.

kunau, 2019-10-11:
The above fix did not work under Catalina, still see this:
[image]

radeksalomon, 2019-10-12:
Did you try to add Full Disk Access to Balena Etcher?
[image]

yabbah, 2019-10-12:
I try burn ubuntu and vmware on 2 different sticks. I always get the error message: "Something went wrong. If it is a compressed image, please check that the archive is not corrupted." 
Can anyone burn images in catalina?

satmandu, 2019-10-12:
> Did you try to grant Full Disk Access to Balena Etcher?
> [image]

Granting full disk access doesn't appear to make a difference.

yabbah, 2019-10-12:
Yes tried that. No difference.

stiucsib86, 2019-10-13:
> FYI Running `sudo /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher` appears to work.

This works. Catalina will prompt for additional permission prior to flashing.

kunau, 2019-10-14:
The above does not work on my system, cannot 'check it for malicious software'.

lurch, 2019-10-14:
> cannot 'check it for malicious software'.

#2911

thundron, 2019-10-19:
Closing as duplicate of #2833 , we have a PR open that will solve issues with macOS Catalina

closed 2019-10-19 by thundron

osxrox, 2019-10-22:
sudo /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher

Worked for me burningan ISO to a USB stick.

maximilliangeorge, 2020-11-30:
> > Did you try to grant Full Disk Access to Balena Etcher?
> > [image]
> 
> Granting full disk access doesn't appear to make a difference.

This worked for me though. Thanks!