balena-io/etcher #2914: Option in settings to disable ad tracking (Revived)

EpicLPer, 2019-09-29:
- **Etcher version:** Latest and ongoing
- **Operating system and architecture:** Any
- **Image flashed:** Doesn't matter
- **Do you see any meaningful error information in the DevTools?** No need to.

This is Take 2 on an issue that has wrongfully been closed and locked now. We have the right to fight for our privacy, if an issue goes offtopic there are mods who are able to steer it into the right direction, no need to lock it directly. The issue in question is #2599 .

Etcher is currently showing tracking ads without the users consent and are also tracking people located in the EU which is against the GDPR, as pointed out by many other users on the old now wrongfully closed issue. They also do not show any warnings about this when showing those ads.

All we ask for is an option to turn this ad-tracking off, nothing more, yet Balena ignores such requests happily as of now. If this issue is once again ignored the GDPR will be informed of this action.

petrosagg, 2019-09-29:
@EpicLPer the issue was closed because the question was answered. The lock happened 16 days later after the issue went offtopic.

But to recap, the "tracking ads" are actually useful projects relevant to the community. We are not interested in removing them as they help both the users and us to continue fund the development of etcher.

> Etcher is currently showing tracking ads without the users consent and are also tracking people located in the EU which is against the GDPR

Collecting anonymous data without consent is fine under GDPR. We have no intention of breaking any law and we have consulted our legal team to ensure that. You can my detailed comment here https://github.com/balena-io/etcher/issues/2766#issuecomment-531437542

closed 2019-09-29 by petrosagg

rradar, 2020-03-07:
This is take 3 @EpicLPer 

Use [usbimager](https://gitlab.com/bztsrc/usbimager) which takes 250kb of your disk (etcher does take almost half a gigabyte with all bloat/ad/spy/xyz-ware) and have the same features (minus tracking/spying/ads...) but flashing your images completely privately now :tada:

zearp, 2020-11-05:
@rradar Thanks for sharing a nice alternative with a GUI.

---

I'm still using the command line version from before the name change. It still works fine today. Funnily enough it's no longer maintained, I guess they want everybody to use the GUI that's slow and bloated and loads up "ads" (for other Balena projects) and opens lots of connections to many servers even with privacy options enabled. It shouldn't even try to connect to Google Analytics with the privacy option enabled. Yet is does and Balena doesn't even see it as an issue. These kind of apps should only connect to the outside world when checking for updates is enabled.

I'm in the process of auditing apps for a local municipality and if apps connect to Google or other analytics services without the option to properly disable this (eg. no more outgoing connections) they're going to be replaced. I see others around me doing the same. I hope it becomes a trend. Apps shouldn't be connecting to all kinds of data harvesting and analytics platforms sharing who knows what about your system with those services. I'd argue that apps shouldn't connect to anything without the user opting-in.

It's probably also a violation of GDPR in European countries, if only for the fact you're not actively opting-in for it, you have to opt-out.

It's telling how little interest Balena has in the privacy of their users. They seem happy so see privacy conscious users leave and move on to different apps so they continue harvesting data of as many people as possible that use this bloated app.