balena-io/etcher #2915: Etcher can't handle Arch Linux archives for Raspberry Pi

RichMorin, 2019-09-29:
- **Etcher version:** Version 1.5.57 (1.5.57)
- **Operating system and architecture:** macOS Version 10.13.6 (17G4015)
- **Image flashed:** ArchLinuxARM-rpi-3-latest.tar.gz
- **Do you see any meaningful error information in the DevTools?** No

I tried using this image, downloaded from the Arch Linux ARM [Downloads](https://archlinuxarm.org/about/downloads) page. I received the following dialog:

> Invalid Image
> .../ArchLinuxARM-rpi-3-latest.tar.gz` is not a supported image type.

Looking around the web, I gather that creating a bootable device for Arch Linux is a rather complicated task. So, it's exactly the sort of problem I'd like Etcher to handle for me. What are the chances of this limitation being resolved?

P.S. I'm not sure how to report this kind of meta-issue, but...
This field's template says:
"You can open DevTools by pressing ... `Cmd+Alt+I` if you're on Mac OS."
The Mac has an Option key, not an Alt key. Also, the name of the operating system is macOS.

lurch, 2019-10-02:
I think that's a "problem" for ArchLinuxARM to solve, not a problem for Etcher to solve.
Etcher is designed to flash raw disk images (with the raw disk image containing partitions, filesystems and files), but ArchLinuxARM (for whatever reason) only provide the files that sit "on top of" the filesystem.

lurch, 2019-10-02:
> P.S. I'm not sure how to report this kind of meta-issue, but...

You could submit a Pull Request against https://github.com/balena-io/etcher/blob/master/.github/ISSUE_TEMPLATE.md

BTW I suspect the Alt / Option key difference probably depends whether you're plugging in a Mac USB keyboard or a PC USB keyboard? :man_shrugging:

RichMorin, 2019-10-02:
> On Oct 2, 2019, at 02:01, Andrew Scheller <notifications@github.com> wrote:
> 
> I think that's a "problem" for ArchLinuxARM to solve, not a problem for Etcher to solve.
> Etcher is designed to flash raw disk images (with the raw disk image containing partitions, filesystems and files), but ArchLinuxARM (for whatever reason) only provide the files that sit "on top of" the filesystem.

I agree that ArchLinuxARM seems to be causing the problem, but I think there's very
little chance that they will be interested in changing their distribution format.
Since the balenaEtcher project seems more inclined to help naive users, I'm hoping
that someone on it will be willing to take up this challenge.

> You could submit a Pull Request against https://github.com/balena-io/etcher/blob/master/.github/ISSUE_TEMPLATE.md

done - https://github.com/balena-io/etcher/pull/2919

> BTW I suspect the Alt / Option key difference probably depends whether you're plugging
> in a Mac USB keyboard or a PC USB keyboard?

Although it's possible to plug a PC USB keyboard into a Mac (and vice versa), the Mac
documentation always uses the term "Opt", so it seems best to be consistent with that.

-r

lurch, 2019-10-02:
> someone on it will be willing to take up this challenge.

I think that if Etcher _wanted_ to be able to flash ArchLinuxARM images, then it'd need to ask the user what size partitions they wanted, what filesystems they wanted those partitions formatted with, etc.? Which is obviously way outside the scope of what Etcher currently does.

Disclaimer: These are only my personal opinions and I obviously have no idea what Balena do or don't have planned for Etcher.

unquietwiki, 2019-10-10:
* https://archlinuxarm.org/platforms/armv8/broadcom/raspberry-pi-3
* https://techcommunity.microsoft.com/t5/Containers/Tar-and-Curl-Come-to-Windows/ba-p/382409

Instructions on Linux at least are to make 2 partitions; one for a FAT32 boot loader; one ext4 to dump the operating system to. Win10 has tar support now; so as long as there's something that talks to ext4 in the toolchain, maybe it can be automated?

lurch, 2019-10-11:
> as long as there's something that talks to ext4 in the toolchain

If you want to do some experimentation yourself, https://github.com/balena-io/node-ext2fs might be a good starting point? :man_shrugging: