balena-io/etcher #2917: Unable to UEFI Boot on Windows with flashed drive

ISHBAR, 2019-10-01:
- **Etcher version:** 1.5.56
- **Operating system and architecture:** macOS 10.14.6
- **Image flashed:** UEFI Iso
- **Do you see any meaningful error information in the DevTools?** N/A

I am encountering an issue where an image is not bootable when made with etcher. In order to get this to work I have to format as MS-DOS / MBR partition scheme. If I mount my ISO and cp the contents to the drive, UEFI boot works.

If flash via etcher, it does not boot. (You select the drive on startup (in Windows) and it goes back to the boot selector).

Is this expected behavior? Could we get the option to flash for a specific destination?
The CLI equivalent of this workflow would be: 

    * diskutil eraseDisk FAT32 UNTITLED MBRFormat /dev/disk#
    * hdiutil mount /Volumes/path_to_iso/my_image.iso
    * cp -rp /Volumes/DVD_ROM/* /Volumes/UNTITLED

lurch, 2019-10-02:
Duplicate of #2704 ?

UAnton, 2021-03-02:
etcher can't write bootable windows iso to USB from MacOS

lurch, 2021-03-04:
@UAnton See https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#why-is-my-drive-not-bootable

maxmod, 2021-05-19:
Use rufus or ventoy

maxmod, 2021-05-19:
if flash drive corrupt use this https://bit.ly/3oBO5JO

maxmod, 2021-05-19:
first remove all partition than use etcher