balena-io/etcher #2920: Persistent error to flash SD cards

barakusT, 2019-10-03:
- version 1.5.57
- Operating system and architecture: OSX 10.15 / Darwin MacBook.local 19.0.0 Darwin Kernel Version 19.0.0: Wed Sep 25 21:13:50 PDT 2019; root:xnu-6153.11.26~4/RELEASE_X86_64 x86_64
- Image flashed: BalenaOS for RusberryPi4 & Ruspbian OS
- Do you see any meaningful error information in the DevTools?   

1- {"stack":"Error: EPERM: operation not permitted, open '/Users/root1/Downloads/balena-cloud-ruspPi-raspberrypi4-64-2.41.0+rev4-dev-v10.2.2.img'","message":"EPERM: operation not permitted, open '/Users/root1/Downloads/balena-cloud-ruspPi-raspberrypi4-64-2.41.0+rev4-dev-v10.2.2.img'","name":"Error","code":"EPERM","syscall":"open","errno":-1,"image":"balena-cloud-ruspPi-raspberrypi4-64-2.41.0+rev4-dev-v10.2.2.img"}
--console.(anonymous function) @ /Applications/balenaEtcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:58

[image]

barakusT, 2019-10-03:
found the problem at issues #2833 
by typing this command -> sudo /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher

zvin, 2019-11-06:
Duplicate of #2833

closed 2019-11-06 by zvin