balena-io/etcher #2921: Many to one drive data collection

konmouz, 2019-10-04:
A user suggested a new feature that basically inverts Etcher's function. Instead of writing from one drive to make, there could be an option to collect data from multiple drives to a single one. This feature is particularly useful on EtcherPro. Comment can be found [here](https://www.balena.io/blog/taming-the-hard-in-hardware-in-8-steps/).

[image]

cc @zvin @thundron

lurch, 2019-10-04:
...but I guess in this case they'd want to read the "data" from the drives (i.e. reading at the filesystem level), rather than making images of the entire drives (i.e. reading at the raw-block level) ?

konmouz, 2019-10-04:
Yes, this is correct.