balena-io/etcher #2926: "balenaEtcher" can't be opened because Apple cannot check it for malicious software

kunau, 2019-10-11:
- **Etcher version:** == 1.5.58

[image]

[image]

- **Operating system and architecture:** 
- **Image flashed:** 
- **Do you see any meaningful error information in the DevTools?**

kunau, 2019-10-11:
OS== OSX Catalina

The application will not start up, so there was no image about to be flashed.
DevTools not available because the application will not start.

benhichak, 2019-10-12:
Seeing the same issue. 

Etcher v1.5.58
MacOS v10.15 (Catalina)

radeksalomon, 2019-10-12:
This is not an issue. You can start the app by right clicking on the app and click Open. This will bypass the Apple's security check and app will start. It works with every app you need to start if it does not pass the security check. This procedure has to be done only during the first start of newly downloaded app.

erikhermesman, 2019-10-12:
On Catalina, I'm having to right click and open twice. The first time tells me it can't be opened, and the second time I do a right click - open, the dialogue box allows me to open the app.

lurch, 2019-10-14:
See also #2833

thundron, 2019-10-19:
Closing as duplicate of #2833 , we have a PR open that will solve issues with macOS Catalina

closed 2019-10-19 by thundron