balena-io/etcher #2927: wont run on (K)ubuntu linux

erikLundstedt, 2019-10-13:
- **Etcher version:** 
balenaEtcher-1.5.58-ia32
- **Operating system and architecture:** 
kubuntu (latest 19.04?)
neofetch gives OS: Ubuntu 19.04 x86_64 
- **Image flashed:** 
none
- **Do you see any meaningful error information?** 
doubbleclicking the appimage does nothing
./balenaEtcher-1.5.58-ia32.AppImage gives the following error:
dlopen(): error loading libfuse.so.2

AppImages require FUSE to run. 
You might still be able to extract the contents of this AppImage 
if you run it with the --appimage-extract option. 
See https://github.com/AppImage/AppImageKit/wiki/FUSE 
for more information

I DO have fuse installed so i dont know why it gives me an error

also:
Subsurface.appimage (provided by the appimage webbsite) runs fine in the same directory with the exact same settings

lurch, 2019-10-14:
It looks like you're trying to run the 32-bit AppImage (balenaEtcher-1.5.58-**ia32**) on a 64-bit OS (Ubuntu 19.04 **x86_64**) ?
Try using the 64-bit AppImage :slightly_smiling_face:

thundron, 2019-10-19:
Seems like what @lurch said, try the 64bit AppImage. Closing for now, let us know if it works!

closed 2019-10-19 by thundron