balena-io/etcher #2928: Ether unable to write ISO file to USB

msnkhan-hub, 2019-10-13:
Ether is unable to write ISO file on USB - I only get an error message "The write process ended unexpectedly"
- Etcher version: 1.5.58 
- Operating system and architecture: MacOS Catalina
- Image flashed: openSUSE-Leap-15.1-DVD-x86_64.iso
- Do you see any meaningful error information in the DevTools? - See below message from the log:

Sun Oct 13 2019 20:46:10 GMT+0100 (BST) Restart after failure ({"applicationSessionUuid":"7eb8466d-5ff3-4502-bde3-91fdb0b94e8e","flashingWorkflowUuid":"19c0e1b6-cf7c-4687-909a-c28abfe61d28","sample":0.1})

lurch, 2019-10-14:
> Operating system and architecture: MacOS Catalina

See #2833

semi72, 2019-10-16:
i have the same problem with an Ubuntu ISO.

Here is my error from DevTools:

/Applications/balenaEtcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:58

{"stack":"Error: EPERM: operation not permitted, open '/Users/cs/Downloads/ubuntu-18.04.3-live-server-amd64.iso'","message":"EPERM: operation not permitted, open '/Users/cs/Downloads/ubuntu-18.04.3-live-server-amd64.iso'","name":"Error","code":"EPERM","syscall":"open","errno":-1,"image":"ubuntu-18.04.3-live-server-amd64.iso"}

console.(anonymous function) @

lurch, 2019-10-17:
@semi72 A lot of people in #2833 are reporting that using `sudo` is an effective workaround
(the problem is with the latest version of macOS, not specifically with ISO files).

thundron, 2019-10-19:
Closing as duplicate of #2833 , we have a PR open that will solve issues with macOS Catalina

closed 2019-10-19 by thundron