balena-io/etcher #2929: After start I'm getting Something went wrong

klaygomes, 2019-10-13:
- **Etcher version:** 
Version 1.5.58 (1.5.58)
- **Operating system and architecture:** 
Mac OS Catanlina - 10.15
- **Image flashed:** 
OpenELEC-RPi2.arm-8.0.4.img.gz
- **Do you see any meaningful error information in the DevTools?** 
{"stack":"Error: EPERM: operation not permitted, open '/Users/cleiton/Downloads/OpenELEC-RPi2.arm-8.0.4.img.gz'","message":"EPERM: operation not permitted, open '/Users/cleiton/Downloads/OpenELEC-RPi2.arm-8.0.4.img.gz'","name":"Error","code":"EPERM","syscall":"open","errno":-1,"image":"OpenELEC-RPi2.arm-8.0.4.img.gz"}

klaygomes, 2019-10-13:
Just a update, I give Etcher full fdisk access (Security & Privacity ->  Full disk access). Now I'm getting:

{"name":"Error","message":"EPERM: operation not permitted, open '/dev/rdisk2'","stack":"Error: EPERM: operation not permitted, open '/dev/rdisk2'","code":"EPERM","syscall":"open","errno":-1},"applicationSessionUuid":"7ff9e459-5d34-43f9-be7f-80d9a73d91c2","flashingWorkflowUuid":"2c490057-7731-4d4f-92a6-da9d1bf64114","flashInstanceUuid":"fcbe870d-95f4-4836-bac8-8a6db964aeb3","image":"/Applications/OpenELEC-RPi2.arm-8.0.4.img.gz","drives":[{"size":62587404288,"isVirtual":false,"enumerator":"DiskArbitration","logicalBlockSize":512,"raw":"/dev/rdisk2","error":null,"isReadOnly":false,"displayName":"/dev/disk2","blockSize":512,"isSCSI":false,"isRemovable":true,"device":"/dev/disk2","busVersion":null,"isSystem":false,"busType":"USB","isCard":false,"isUSB":true,"devicePath":"IODeviceTree:/PCI0@0/PEG2@1,2/UPSB@0/DSB2@2/XHC3@0","mountpoints":[],"description":"Generic MassStorageClass Media","isUAS":null}],"driveCount":1,"uuid":"fcbe870d-95f4-4836-bac8-8a6db964aeb3","unmountOnSuccess":true,"validateWriteOnSuccess":true,"trim":true,"sample":0.1})

I tried using `dd` and it worked fine, so I assume the problem is with Etcher.

lurch, 2019-10-14:
See #2833

thundron, 2019-10-19:
Closing as duplicate of #2833 , we have a PR open that will solve issues with macOS Catalina

closed 2019-10-19 by thundron