balena-io/etcher #2930: maybe broken balenaEtcher-1.5.58.dmg

enkunkun, 2019-10-14:
`.dmg` file only. `.zip` file version starts normally.

- **Etcher version:** 1.5.58
- **Operating system and architecture:** MacOS Mojave (10.14.6)
- **Image flashed:** None
- **Do you see any meaningful error information in the DevTools?** None

```
$ hdiutil attach -verbose balenaEtcher-1.5.58.dmg
Initializing…
DIBackingStoreInstantiatorProbe: interface  0, score      100, CBSDBackingStore
DIBackingStoreInstantiatorProbe: interface  1, score    -1000, CBundleBackingStore
DIBackingStoreInstantiatorProbe: interface  2, score    -1000, CRAMBackingStore
DIBackingStoreInstantiatorProbe: interface  3, score      100, CCarbonBackingStore
DIBackingStoreInstantiatorProbe: interface  4, score    -1000, CDevBackingStore
DIBackingStoreInstantiatorProbe: interface  5, score    -1000, CCURLBackingStore
DIBackingStoreInstantiatorProbe: interface  6, score    -1000, CVectoredBackingStore
DIBackingStoreInstantiatorProbe: interface  0, score      100, CBSDBackingStore
DIBackingStoreInstantiatorProbe: interface  1, score    -1000, CBundleBackingStore
DIBackingStoreInstantiatorProbe: interface  2, score    -1000, CRAMBackingStore
DIBackingStoreInstantiatorProbe: interface  3, score      100, CCarbonBackingStore
DIBackingStoreInstantiatorProbe: interface  4, score    -1000, CDevBackingStore
DIBackingStoreInstantiatorProbe: interface  5, score    -1000, CCURLBackingStore
DIBackingStoreInstantiatorProbe: interface  6, score    -1000, CVectoredBackingStore
DIFileEncodingInstantiatorProbe: interface  0, score    -1000, CMacBinaryEncoding
DIFileEncodingInstantiatorProbe: interface  1, score    -1000, CAppleSingleEncoding
DIFileEncodingInstantiatorProbe: interface  2, score    -1000, CEncryptedEncoding
DIFileEncodingInstantiatorProbe: interface  0, score      900, CUDIFEncoding
DIFileEncodingNewWithBackingStore: CUDIFEncoding
DIFileEncodingNewWithBackingStore: instantiator returned 0
DIFileEncodingInstantiatorProbe: interface  0, score    -1000, CSegmentedNDIFEncoding
DIFileEncodingInstantiatorProbe: interface  1, score    -1000, CSegmentedUDIFEncoding
DIFileEncodingInstantiatorProbe: interface  2, score    -1000, CSegmentedUDIFRawEncoding
DIDiskImageInstantiatorProbe: interface  0, score     1000, CUDIFDiskImage
DIDiskImageInstantiatorProbe: interface  1, score        0, CSparseBundleDiskImage
DIDiskImageInstantiatorProbe: interface  2, score        0, CSparseDiskImage
CRawDiskImage: data fork length 0x0000000005019D7C (83991932) not a multiple of 512.
DIDiskImageInstantiatorProbe: interface  3, score     -100, CRawDiskImage
DIDiskImageInstantiatorProbe: interface  4, score        0, CDARTDiskImage
DIDiskImageInstantiatorProbe: interface  5, score        0, CDiskCopy42DiskImage
DIDiskImageInstantiatorProbe: interface  6, score    -1000, CNDIFDiskImage
DIDiskImageInstantiatorProbe: interface  8, score     -100, CShadowedDiskImage
DIDiskImageInstantiatorProbe: interface  9, score    -1000, CCFPlugInDiskImage
DIDiskImageInstantiatorProbe: interface 10, score     -100, CWrappedDiskImage
DIDiskImageNewWithBackingStore: CUDIFDiskImage
DIDiskImageNewWithBackingStore: instantiator returned 0
Verifying…
Verification completed…
Error 0 (Undefined error: 0).
expected   CRC32 $0D403621
Attaching…
Mounting…
2019-10-14 15:06:48.396 diskimages-helper[8040:7559226] DIHelperHDID serveImage: attaching drive
{
    autodiskmount = 1;
    "hdiagent-drive-identifier" = "817D3F56-3DD2-4D3B-A235-CC77116A04F5";
    "unmount-timeout" = 0;
}
2019-10-14 15:06:48.398 diskimages-helper[8040:7559226] DIHelperHDID serveImage: connecting to myDrive 0x3F0F
2019-10-14 15:06:48.398 diskimages-helper[8040:7559226] DIHelperHDID serveImage: register _readBuffer 0x10c478000
2019-10-14 15:06:48.398 diskimages-helper[8040:7559226] DIHelperHDID serveImage: activating drive port 16651
2019-10-14 15:06:48.399 diskimages-helper[8040:7559226] DIHelperHDID serveImage: set cache enabled=TRUE returned SUCCESS.
2019-10-14 15:06:48.399 diskimages-helper[8040:7559226] DIHelperHDID serveImage: set on IO thread=TRUE returned SUCCESS.
2019-10-14 15:06:48.399 diskimages-helper[8040:7559226] -processKernelRequest: will sleep received
2019-10-14 15:06:48.555 diskimages-helper[8040:7559223] _postflightMountPointsAfterDAMount: disk4s1 aborting because no mount point found.
2019-10-14 15:06:48.556 diskimages-helper[8040:7559223] _postflightMountPointsAfterDAMount: disk4 aborting because no mount point found.
2019-10-14 15:06:48.556 diskimages-helper[8040:7559223] allowMountReturningDictionary: detaching because no mountable filesystems.
Attaching…
Error 112 (no mountable file systems).
Finishing…
DIHLDiskImageAttach() returned 112
hdiutil: attach failed - no mountable file systems
```

adambkaplan, 2020-04-30:
This appears to be an issue on Catalina as well with the 1.5.81 release

thundron, 2020-05-04:
Does that happen if you open the dmg without using hdiutil? (i.e. with a double click)

enkunkun, 2020-05-05:
When I created the issue, it also happened with a double click.
I tried it with the latest version (v1.5.83) and it successfully opened the .dmg file (on macOS Mojave 10.14.6).

I'm not sure when it was fixed, but I'll close the issue.
Thank you!

closed 2020-05-05 by enkunkun