balena-io/etcher #2933: Etcher not opening up burn images on Catalina

LazyHRanch, 2019-10-14:
Recently upgraded to macOS Catalina and found Etcher not working. I get an error: 
"Something Went wrong. If it is a compressed image, please check that the archive is not corrupted."

I know it's not the image because I'm using the same images that I used in the previous MacOS version and I verified it on a PC that the image is OK. I also tried a couple of different images to make sure it wasn't related to a specific image.

I noticed that I had to "Open" Etcher twice after installing to get through the warnings but then it opened up just fine. Amy suggestions or is anyone else seeing this error?

lurch, 2019-10-14:
Duplicate of #2833 ?

LazyHRanch, 2019-10-14:
It looks like that issue is with opening Etcher directly. I can get Etcher to run but when I try to write to the SD card is when I'm getting the above error. And yes it does ask for permissions when starting the burn process.
[image]

lurch, 2019-10-14:
Sorry if I didn't understand your issue. I don't have a Mac myself, I was just pointing out that there are already "known problems" with macOS Catalina. Perhaps your error is indeed different.

roura356a, 2019-10-15:
Exactly the same happening here. Once you give permissions to the app to be executed, you go to flash any image and... fails with above screenshot message.

taiansu, 2019-10-15:
I can confirm that this issue is a duplicate of #2833, which just also provides the error message in DevTools. In the thread, there is a temporary solution by launching etcher using CLI.
```
sudo /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher
```

That solves the issue of the screenshot on this issue.

SebiWolze, 2019-10-15:
> I can confirm that this issue is a duplicate of #2833, which just also provides the error message in DevTools. In the thread, there is a temporary solution by launching etcher using CLI.
> 
> ```
> sudo /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher
> ```
> 
> That solves the issue of the screenshot on this issue, on my machine, at least.

That worked for me. 
Before I had the same issue like on the screenshots before.

AegerBorder, 2019-10-16:
> That worked for me.
> Before I had the same issue like on the screenshots before.

Yep. Another confirmation here. Thanks for this workaround.

majkassab, 2019-10-16:
> I can confirm that this issue is a duplicate of #2833, which just also provides the error message in DevTools. In the thread, there is a temporary solution by launching etcher using CLI.
> 
> ```
> sudo /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher
> ```
> 
> That solves the error showing the screenshot on this issue, on my machine, at least.

That worked for me. thx

thundron, 2019-10-19:
Closing as duplicate of #2833 , we have a PR open that will solve issues with macOS Catalina.
@stephendwolff Yes, we need to find the right entitlements for Etcher, as USB devices + a couple more don't seem to suffice for now

closed 2019-10-19 by thundron

AdvaitT17, 2019-12-10:
Worked for me as well! Thanks

Aculisme, 2020-01-11:
Worked here too. Thanks!

ygreq, 2020-02-07:
Can this bug be fixed, please? 
The terminal command worked for me as well, btw

lurch, 2020-02-07:
@ygreq I don't have a Mac myself, but from what I've read elsewhere I thought this bug was fixed? :man_shrugging: 
What version of Etcher are you using?

ygreq, 2020-02-07:
I was using 1.5.59.  I just updated to 1.5.76. And it works! Thank you for the heads up, lurch!!

LanceHaverkamp, 2020-03-20:
As of March 2020, this is still not fixed in Linux. It asks for the root password, then fails.

lurch, 2020-03-20:
@LanceHaverkamp I suggest you open a new issue and describe your problem in as much detail as possible.

StefanoSGI, 2020-05-03:
+1 same issue reported in https://github.com/balena-io/etcher/issues/2933#issuecomment-541833707 still happening on Mac OS Catalina 10.15.4 with Etcher v.1.5.83 trying to flash `2020-02-13-raspbian-buster-lite.img`. 
- Pick image
- Select target
- Click `Flash!`
- Enter password to authorize app
- This appears:
[image]

lurch, 2020-05-04:
@StefanoSGI Are you running on Catalina as a "non-admin" user? If so, does [this](https://github.com/balena-io/etcher/issues/3107#issuecomment-602047807) summarise what you're experiencing?

StefanoSGI, 2020-05-04:
@lurch the account I'm using is an Admin on my laptop.

stevyhacker, 2020-06-08:
@StefanoSGI still happening, the same scenario with ubuntu images and its derivatives images on Mac OS Catalina latest and admin user account.

stevyhacker, 2020-06-08:
Just tasted the latest build 1.5.95 on Mac OS and it works fine now 😁

ph00lt0, 2020-07-09:
Unfortunately I have the same issue. Just installed lastest version, nothing seems to help, neither starting it through terminal.

davidmgrantham, 2020-07-15:
I having the same issue on build 1.5.101.
OS: MacOS Catalina 10.15.5

tomaspiaggio, 2020-09-03:
I am also experiencing the issue. I am running a Macbook Pro 15 on Catalina (10.15.6). I have been flashing lots of images these last two weeks and I experienced this issue multiple times, with multiple different images (from the top of my memory fedora 32, Debian 10). These are the logs I am getting:

[image]
I did not get into reading the code, but this might help somebody.

sscotth, 2020-09-08:
Same issue `Version 1.5.106`

tazdrum73, 2020-09-18:
Catalina 10.15.6
BalenaEtcher 1.5.109

Same issue. Tried sudo and it did not work, but if I switch to root and start the app as root from cli it works.

hahahaharada, 2020-09-29:
Yosemite OS

U need format flashdrive (FAT system) before flash distro. 
Otherwise it will give the same error... This is the trick! 

tested with and without sudo... 
sudo /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher

canehusker, 2020-10-14:
> That worked for me.
> Before I had the same issue like on the screenshots before.

Yes, that worked. Thank you. Running it from Mac (Catalina 10.15.6) kept failing.

HelenGuohx, 2020-10-16:
I u**ncompressed the file in terminal to get the image**. And The problem solved. 
My OS version is  macOS mojave 10.14.6
Etcher version is Version 1.5.109 (1.5.109)

SamuelMiller, 2020-10-18:
Hello, I intermittently get the same error even when running with sudo privileges through the terminal in Catalina--at least half the time or more. Sometimes, restarting my computer solves it, sometimes not.

sebastianfelipe, 2020-10-28:
> I can confirm that this issue is a duplicate of #2833, which just also provides the error message in DevTools. In the thread, there is a temporary solution by launching etcher using CLI.
>
> ```
> sudo [...]

This man is a genius!

PierreScerri, 2020-11-04:
I have done it as follows (on MacOS Catalina 10.15.3):

Launch Disk Utility
Format destination drive (FAT32 - GUID)
Unmount the drive.
Launch Balena Etcher normally.
Flash normally.

I have followed this sequence on 2 drives that were giving the error.

Hope it helps

AlexanderJGomez, 2020-11-21:
> I have done it as follows (on MacOS Catalina 10.15.3):
> 
> Launch Disk Utility
> Format destination drive (FAT32 - GUID)
> Unmount the drive.
> Launch Balena Etcher normally.
> Flash normally.
> 
> I have followed this sequence on 2 drives that were giving the error.
> 
> Hope it helps

Tried running with `sudo` command but did not work. Unmounting the drive fixed it for me, thanks!

almolinagithub, 2020-11-22:
The above worked for me too.

maslaral, 2020-11-25:
The above worked for me. Except in addition to unmounting the drive, I had to launch Etcher via the command line with sudo.

mhazley, 2020-11-27:
@AlexanderJGomez cheers - this worked!

raindog308, 2020-12-01:
Have tried all fixes here - sudo, running as root, reformatting as GUID, yet the same issue (same error) persists.

This is Etcher 1.5.111 on macOS 10.15.6 Catalina.

PierreScerri, 2020-12-01:
Did you unmount the drive before using Balena?

raindog308, 2020-12-01:
If I unmount it, it does not show up in the Select Target.  If I mount it (just insert the card and it automounts), it shows up as "Apple SDXC Reader Media" on /dev/disk4.

raindog308, 2020-12-01:
I see the issue...in Finder, you can only eject the volume (e.g., "UNTITLED") which also ejects the reader.  In Disk Utility, it's possible to eject the volume and still leave the reader mounted.  When I do that, it works.

tszumowski, 2020-12-01:
@raindog308 I literally was just about the write the same thing haha. 
Hitting "Unmount" here did it for me. I didn't have to run as sudo either:

[image]

PierreScerri, 2020-12-01:
Yes, as I said previously, just unmount the drive in disk utility and use Balena to write the image.

tszumowski, 2020-12-01:
> Yes, as I said previously, just unmount the drive in disk utility and use Balena to write the image.

Whoops! I missed your post in the mix of the others. Yes, your steps are perfect

WorldWideWebDev, 2021-04-05:
On your Mac, Go to  Firewall/ Privacy/ scroll down to Files and Folders, Ensure "Balena Etcher " Has access to "Removable Volumes" by ticking the box.

PierreScerri, 2021-04-05:
@WorldWideWebDev Balena is not listed under Files and Folders on my Mac.  Cannot add it as the '+' is disabled even with the padlock unlocked.