balena-io/etcher #2936: Windows 10 blank window - error in console

askovi, 2019-10-16:
- **Etcher version: 1.5.59** 
- **Operating system and architecture: Window 10 x64** 
- **Image flashed: n/a** 
- **Do you see any meaningful error information in the DevTools?** 
[image]

askovi, 2019-10-18:
This is not working for me anymore on Windows 10 at all. Now it's just a blank window. 

[image]

lurch, 2019-10-18:
If even older versions are no longer working (when they used to be fine), perhaps the problem is with Windows 10 rather than with Etcher?
IIRC other users have reported that sometimes various "security features" get automatically enabled in Windows, which prevents Etcher running? Maybe you'll be able to find some useful advice somewhere in https://github.com/balena-io/etcher/search?o=desc&q=windows+10&s=updated&type=Issues ? :man_shrugging:

bpp9614, 2020-04-08:
@askovi - Could you find a solution? I am having similar issue on Windows 10. 

[image]

askovi, 2020-04-15:
> @askovi - Could you find a solution? I am having similar issue on Windows 10. 
> 
> 
> 
> [image]
> 
> 

It just started working again after updating to the latest version. Not sure what the issue was...

lurch, 2020-04-15:
@bpp9614 From your error-log, it looks like you're trying to run a locally-built-from-source version of Etcher? Does a version downloaded from https://www.balena.io/etcher/ work okay?

bpp9614, 2020-04-19:
Yes the downloaded one work OK. I think its related to the npm version I
was using. Got fixed after using right version of npm.

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

closed 2023-07-12 by dfunckt