balena-io/etcher #2938: Download speed VERY VERY SLOW at China!!!!!

ScottLiu88, 2019-10-18:
- **Etcher version:** 
- **Operating system and architecture:** 
- **Image flashed:** 
- **Do you see any meaningful error information in the DevTools?**

lurch, 2019-10-18:
See #2083 and #2336 and #2910

mifi, 2019-10-26:
Same problem in singapore

thundron, 2019-12-11:
We'll do our best to reduce the package size (e.g. [here](https://github.com/balena-io/etcher/pull/2987)) but for now there's little we can do other than that

closed 2019-12-11 by thundron

lurch, 2019-12-16:
@thundron Just asking out of curiosity, but there doesn't seem to be much change in download-size between 1.5.69 and 1.5.70 ? https://github.com/balena-io/etcher/releases

Antony-L, 2019-12-27:
For you guys in China copy the link for your file and use Xulei(迅雷)