balena-io/etcher #2940: Once you use etcher, you never more use your stick.

runawaydevil, 2019-10-18:
I'm using the last version of etcher to install my linux.
Now I can't format the usb stick, it's awful. I can't do this using diskpart, rufus, sd formatter, gparted, parted and nothing works for me.
I hate this software. I have a pen drive with 64 gb and now only shows a FAT16 device with 1.8gb and a .iso file that cannot be deleted. 

wtfffffff

someone has any idea what can i do?

AngledLuffa, 2019-10-18:
I'm having the exact same problem.  I used Etcher on Windows to make a Centos 7 install stick a few months back.  Now I'd like to make a Centos 8 install stick, and I can't format the old stick in either Windows or Linux, and I can't use Etcher to overwrite the existing stick.

Any guidance on how to avoid this happening would be great.

runawaydevil, 2019-10-18:
> I'm having the exact same problem. I used Etcher on Windows to make a Centos 7 install stick a few months back. Now I'd like to make a Centos 8 install stick, and I can't format the old stick in [...]

The worst thing I did was install this program.

AngledLuffa, 2019-10-18:
I wouldn't go that far.  It permanently turned a $10 USB stick into a Centos 7 install stick... not the worst thing that's ever happened to me.  

I would definitely appreciate it if I could at least turn that stick into a Centos 8 install stick, though

curcuz, 2019-10-19:
@AngledLuffa @runawaydevil there are clear guidelines on how to report issues here that help the maintainers understand and eventually fix an issue. In your case you provided 0 context, no reproduction steps for investigation, and just dumped a random rant here. Can you please elaborate on your issue, attach logs and screenshots, and help us help you?

Also, please read first the following:

https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#flashing-ubuntu-isos

https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#recovering-broken-drives

chrschmitz, 2019-10-19:
These are the steps for Mac users "to recover" USB sticks, which Disk Utility is not able to format anymore.

Open Terminal and look for the identifier of your USB stick:

`diskutil list`

Then enter the following command:

`diskutil eraseDisk FILE_SYSTEM DISK_NAME DISK_IDENTIFIER`

For example, if the name of the device is "/dev/disk2", the file system should be "FAT32" and the name should be "UNTITLED" enter:

`sudo diskutil eraseDisk FAT32 UNTITLED MBRFormat /dev/disk2`

After you enter your administrator password the USB stick will be formatted and will show up in the Finder again.

It is absolutely critical that you use proper syntax when identifying the disk to erase. Improper identification may lead to erasing the wrong disk, permanently destroying any data on it. Do not screw this up!

Hope that helps to "resuscitate" sticks, which are not readable anymore.

Best,
Chris

runawaydevil, 2019-10-19:
> @AngledLuffa @runawaydevil there are clear guidelines on how to report issues here that help the maintainers understand and eventually fix an issue. In your case you provided 0 context, no [...]

I think you should read the thread before, all this bullshit we tried already.

runawaydevil, 2019-10-19:
> These are the steps for Mac users "to recover" USB sticks, which Disk Utility is not able to format anymore.
>
> Open Terminal and look for the identifier of your USB stick:
>
> `diskutil list`
>
> Then [...]

sadly i don't have macOS :(((

curcuz, 2019-10-19:
> I think you should read the thread before, all this bullshit we tried already.

@runawaydevil you are still not providing logs, screenshots or anything we could use to understand a problem that, after "reading your thread" is not illustrated at all. On top of that, you are having a very bad attitude, which also does not help us helping you.

That being said, etcher does not destroy any USB medium, as the only action it performs is flashing disk images **along with their partition table** to the target medium. 
Most likely you have your USB stick actually working but containing partitions under a filesystem your host OS does not support. But we cannot tell, because you are refusing to provide any useful information about the issue you are facing.

AngledLuffa, 2019-10-19:
Please excuse me for piling on with any negativity.  To clarify my initial comment, these are the steps I tried doing:

- download centos 7 image (I forget exactly which)
- use balena etcher to create a centos install thumb drive
- install centos on a desktop (this step is probably not necessary)
- try to format, repartition, etc my thumb drive so I can use it as a thumb drive again.  nothing I try works

This includes diskpart, sadly.  I run "clean" and it says "DiskPart has encountered an error: Access is denied."  There's no physical lock mechanism on the drive, so I'm not sure how that happens.  The error in the system event log is

"Cannot zero sectors on disk \\?\PhysicalDrive1. Error code: 5@0101000F"

If you send me a PM with your address, I can send you the broken thumb drive if that will help the investigation.

runawaydevil, 2019-10-19:
I'll provide screenshots tonight from DISKPART tonight, but remember I tried RUFUS, GPARTED, PARTED and others, including the own etcher to try to overwrite with another OS.

AngledLuffa, 2019-10-19:
Also, to further clarify what tools I was using, this was a few months ago
so I had used version 51.  Haven't tried 59 yet.  Windows 10.

AMDphreak, 2019-10-19:
I think this is related to the partition table bug that I have done extensive testing on.

AngledLuffa, 2019-10-19:
I can give that a try on Linux.  Is there a chance of a workaround or fix for Windows?

AMDphreak, 2019-10-19:
> I can give that a try on Linux. Is there a chance of a workaround or fix for Windows?

Actually, there might be a really dirty hack. If [diskpart's clean command](https://winaero.com/blog/securely-wipe-disk-diskpart-windows-10/) doesn't work, use [dd for windows](http://www.chrysocome.net/dd). dd copies directly to the disk, so find a giant file like 4 GB in size and copy it with dd (data duplicator/disk destroyer):
```
dd if=C:/path/bigfilehere of=<output drive>
```
I'm not sure what the syntax will be like for the output drive on Windows. If the disk is messed up, you might have to use a disk identifier that isn't like "C:" or "F:".

runawaydevil, 2019-10-20:
> @runawaydevil you are still not providing logs, screenshots or anything we could use to understand a problem that, after "reading your thread" is not illustrated at all. On top of that, you are [...]

How I said, I tried everything, here's an example using diskpart like the documentation suggest.

[image]

AngledLuffa, 2019-10-21:
Thanks to your suggestions, I was able to use `wipefs -a /dev/sda` to put the drive back in a state where I can install a new .iso.  Thanks for the assistance.

runawaydevil, 2019-10-21:
> Thanks to your suggestions, I was able to use `wipefs -a /dev/sda` to put the drive back in a state where I can install a new .iso. Thanks for the assistance.

I'll try too.

AngledLuffa, 2019-10-21:
After inserting the USB, I needed to run `umount /dev/sda*` before running `wipefs`

runawaydevil, 2019-10-23:
> After inserting the USB, I needed to run `umount /dev/sda*` before running `wipefs`

I did all and nothing works. When I iused wipefs it gave many errors :(

HEY GUYS ANY SUGGESTION?

Davidnet, 2019-10-23:
@runawaydevil Would mind to show ANY LOGS so the team could help you out, or at least show the errors shown?

AMDphreak, 2019-10-26:
@runawaydevil Your screenshot is in Spanish, but based on my limited Spanish, it seems like diskpart is saying you don’t have sufficient permissions to clean the disk. You need to run diskpart from an Admin command line: start>search “command”> right click and select “run as admin”. Then do the commands.

Also, you said you have done all of the commands and nothing worked. What Operating System and what command line are you running when you used those commands? The commands are for GNU-Linux OSes, but you seem to be exclusively using Windows.

runawaydevil, 2019-10-26:
> @runawaydevil Your screenshot is in Spanish, but based on my limited Spanish, it seems like diskpart is saying you don’t have sufficient permissions to clean the disk. You need to run diskpart from an Admin command line: start>search “command”> right click and select “run as admin”. Then do the commands.
> 
> Also, you said you have done all of the commands and nothing worked. What Operating System and what command line are you running when you used those commands? The commands are for GNU-Linux OSes, but you seem to be exclusively using Windows.
> 
> If you are running wipefs on a Linux machine, make sure you use “sudo” before the command to run it with admin privileges.

hey, thanks for your answer! it's portuguese :D which command worked for you?

AMDphreak, 2019-10-27:
I need you to answer the questions first.
1. Which operating system (OS) are you running when you execute the
commands?
2. If you are using Windows, are you running the commands in an emulated
Linux (GNU-Linux) environment?
3. If you are running a Linux Installer image (from CD or USB flash drive)
or a full working copy of Linux, what is the name of the Linux distribution
(also called a “distro”). This is the name of the project, for example
“Linux Mint” or “Ubuntu” or “Red Hat” or “Fedora” or “Debian”.

runawaydevil, 2019-10-27:
> I need you to answer the questions first. 1. Which operating system (OS) are you running when you execute the commands? 2. If you are using Windows, are you running the commands in an emulated Linux (GNU-Linux) environment? 3. If you are running a Linux Installer image (from CD or USB flash drive) or a full working copy of Linux, what is the name of the Linux distribution (also called a “distro”). This is the name of the project, for example “Linux Mint” or “Ubuntu” or “Red Hat” or “Fedora” or “Debian”.
> -- Ryan Johnson Free Software Foundation <https://www.fsf.org/> Electronic Frontier Foundation <https://www.eff.org/>

1. Which operating system (OS) are you running when you execute the
commands? **Kali Linux**
2. If you are using Windows, are you running the commands in an emulated
Linux (GNU-Linux) environment? **No**
3. If you are running a Linux Installer image (from CD or USB flash drive)
or a full working copy of Linux, what is the name of the Linux distribution
(also called a “distro”). This is the name of the project, for example
“Linux Mint” or “Ubuntu” or “Red Hat” or “Fedora” or “Debian”: **Full copy, Kali Linux**

Bill-Lo3, 2019-10-28:
Program is giving me issues in the Package Manager on Parrot Sec (ill post in another thread on this issue). It did burn the iso to the usb.
After test installing a differ linux version on a laptop, 

Here is how cleared and reformatted the usb to be usable again for any files. 
Open the terminal:

sudo su 
fdisk -l                 (Find your flash drive name)
umount /dev/sdc         (sdc= my usb drive, always check fdisk 1st)
shred -vf -n 1 /dev/sdc    (1 = # of times to shred, shred 1-3 times?)
parted /dev/sdc 
mktable msdos 
y
mkpart primary 0% 100% 
quit 
mkfs.msdos -F 32 /dev/sdc1

AMDphreak, 2019-10-29:
> **shred -vf -n 1 /dev/sdc** (1 = # of times to shred, shred 1-3 times?)

I had no idea there was a function to generate random data to a file (device). That's really useful.

> **parted /dev/sdc**
> **mktable msdos**
> **y**
> **mkpart primary 0% 100%**
> **quit**
> **mkfs.msdos -F 32 /dev/sdc1**

You reformatted as an msdos partition with MBR (master boot record) table. For anyone trying to format as something else, leave off everything starting at `mktable msdos` and replace it with your preferred partitioning tool (command or GUI) or tell parted to format as a different type: EFI + ext4 or EFI + btrfs for example.

AMDphreak, 2019-10-29:
> 1. Which operating system (OS) are you running when you execute the
>        commands? **Kali Linux**
>
>     2. If you are using Windows, are you running the commands in an emulated
>        Linux [...]

I don't use Kali Linux (yet). Kali is based on Debian, so it uses the `apt` package manager. Considering that every other distro includes these commands by default, you should have them installed. If you don't, try installing them:

```
sudo apt-get install parted
```
```
sudo apt-get install fdisk
```

The `wipefs` utility is included by default in the OS and is not available itself as a separate installable package (at least in Linux Mint). It may be available whenever `parted` or `fdisk` is installed (if parted depends on fdisk for functionality).

This is fairly basic stuff at this point. Not sure I have time to cover anything more basic, so do yourself a favor and read up on the command line in Linuxes and the concept of package installers (dpkg vs rpm) and managers (apt vs (yum, dnf)). 

https://www.wikiwand.com/en/Dpkg

lurch, 2019-10-29:
On Ubuntu, the `wipefs` manpage says "The  wipefs  command is part of the util-linux package".

wb7odyfred, 2019-11-14:
https://how-to.fandom.com/wiki/How_to_wipe_a_hard_drive_clean_in_Linux
Wiping the Master boot record (MBR)Edit
If you messed up your master boot record (MBR) you can wipe it using this command : 
dd if=/dev/zero of=/dev/hdX bs=446 count=1 *replace X with the target drive letter

wb7odyfred, 2019-11-14:
if=/dev/zero of=/dev/hdX bs=512 count=1   
Overwrite the Master Boot Record code and the partition table record data  with zeros
Other helpful commands
man dd  or info dd
man od  of info od
sudo fdisk -l   look for all partition tables on the computer
sudo dd if=/dev/sdX bs=512 count =1 | od =Ax -t x1 -a   output the contents of the partition table for disk sdX  change X to select which disk drive   [ a b c d ]   od --help | less
https://www.linuxquestions.org/questions/linux-newbie-8/using-dd-to-zero-the-mbr-query-606489/
https://www.linuxquestions.org/linux/answers/Applications_GUI_Multimedia/How_To_Do_Eveything_With_DD

lurch, 2019-11-15:
@wb7odyfred as already mentioned at https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#gnulinux :wink:

closed 2019-12-11 by thundron