balena-io/etcher #2942: Partition table contains overlapping partitions

AMDphreak, 2019-10-19:
- **Etcher version: 1.5.59** 
- **Operating system and architecture: Windows 10 Pro x64** 
- **Image flashed: Linux Mint 19.2, codename "Tina"** 
- **Do you see any meaningful error information in the DevTools? Don't know. Didn't check.** 

# Problem

Burning ISO files results in an invalid partition table on a USB flash drive. After burning, it appears that the EFI boot partition is after the main partition:

[image]

# MacOS
The issue cannot be automatically fixed by most tools. MacOS fails to fix this. MacOS can reformat the EFI partition, but it can't erase the bad partition (or its entry in the partition table). It reformats as an ISO9660 (Joliet) filesystem (even though the user cannot provide this filesystem type). User must provide a filesystem type in MacOS, which makes it impossible to simply reformat blank, and deleting from the partition table fails.

# Linux
Exploring the file system in Linux has massive problems. sfdisk does not like the partitions. fdisk does not like the partitions. gparted (parted) does not like the partitions.

parted offers to "fix" the partition table by extending a partition (doesn't specify which). Once I ignore it, it reveals there is a problem with the partition table: `Error: Can't have overlapping partitions`.

```bash
mint@mint:~$ sudo parted /dev/sdc
GNU Parted 3.2
Using /dev/sdc
Welcome to GNU Parted! Type 'help' to view a list of commands.
(parted) print                                                            
Warning: Not all of the space available to /dev/sdc appears to be used, you can fix the GPT to use all of the space (an extra
26451456 blocks) or continue with the current setting? 
Fix/Ignore? i                                                             
Error: Can't have overlapping partitions.
Ignore/Cancel? i                                                          
Model: SanDisk Ultra (scsi)
Disk /dev/sdc: 15.6GB
Sector size (logical/physical): 512B/512B
Partition Table: gpt
Disk Flags: pmbr_boot

Number  Start  End     Size    File system  Name        Flags
 1      0.00B  2009MB  2009MB               ISOHybrid   hidden, msftdata
 2      334kB  2726kB  2392kB               ISOHybrid1  hidden, msftdata
```

Getting rid of this was a major chore. I did not want to brick the disk's partition table or wipe the entire disk (ended up doing this anyway), so I tried to edit the partition table using sfdisk:

Port the table out of the disk with sfdisk dump:
```bash
sfdisk -d /dev/sdc > ~/table.txt
```
(don't forget to use `sudo su` to become root before running these, or use `sudo` before the command: `sudo <command>`)

Edit the file, deleting EFI entry:
```bash
nano table.txt
```
or use another better text editor (gedit, kate).

Load the file back into the partition table using sfdisk:
```bash
sfdisk /dev/sdc < ~/table.txt
```

This fails to overwrite the partition table with the correct values. sfdisk does not report errors, but inspecting it afterward reveals it failed.
```bash
sfdisk -d /dev/sdc
[details of partition table...]
```

Attempting to delete the partition or reformat the partition with fdisk or parted fails, because it consults the table first and gets confused.
```bash
fdisk --wipe-partition /dev/sdc
```
```bashfdisk
parted /dev/sdc
Using /dev/sdc
(parted) rm 2
```

# Fix (in Linux only):
Unfortunately, there is no fix in MacOS short of searching for a different drive writer online or a command line util to wipe the drive. `wipefs` might be available, but no guarantees.

Fix in Linux involves wiping the entire drive:
```
wipefs /dev/sdc
```
For FAT, ZFS, or GPT, use the -a switch with wipefs to make sure all bits (magic bits) are erased:
```
sfdisk --wipe -a
```
from its manual page (`man sfdisk`):
> ```
>        Note that some filesystems and some partition tables store  more  magic
>        strings  on the device (e.g. FAT, ZFS, GPT).  The wipefs command (since
>        v2.31) lists all the offset where a magic strings have been detected.
> 
>        When option -a is used, all magic strings that are visible for libblkid
>        are  erased.  In this case the wipefs scans the device again after each
>        modification (erase) until no magic string is found.
> ```
## Other options
`wipefs` may just be a wrapper for a call to sfdisk:
```
sfdisk --wipe /dev/sdc
```
fdisk also may be a wrapper for a call to sfdisk:
```
fdisk --wipe /dev/sdc
```

Reference to other bug reports:
#2940

lurch, 2019-10-19:
AFAICT, this is just a more detailed explanation of the "problem" mentioned [here](https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#flashing-ubuntu-isos).

And I think https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#gnulinux is a simpler "fix" ?

AMDphreak, 2019-10-19:
> AFAICT, this is just a more detailed explanation of the "problem" mentioned [here](https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#flashing-ubuntu-isos).
> 
> And I think https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#gnulinux is a simpler "fix" ?

That's a valuable differential diagnosis. I can't tell. The error messages are different. Mine doesn't seem to be throwing any errors about invalid fake MSDOS partitions.

AngledLuffa, 2019-10-21:
I don't know what sfdisk you're using, but `sfdisk from util-linux 2.23.2` doesn't have --wipe or -a as options

AngledLuffa, 2019-10-21:
However, using `wipefs -a /dev/sda` got the disk back to a state where I can use etcher to flash a new .iso onto the drive.  Good enough for me

thundron, 2019-10-22:
The "issue" here is more the fact that we don't apply any logic over the image being flashed which means whatever is in the image is read and written without any modification. If the partition table is invalid, it is such in the image itself and we decided not to mess with partitions except for specific cases (on Windows, we hold the first bytes to prevent the OS from mounting the target before it's fully flashed).
We do appreciate the thorough analysis anyway and I guess you saw by yourself *why* we don't modify images - because it would be a chore to do so for every image Etcher users flash (us included!), as each and every of them can be and often is different in ways that (always *for now*) we chose not to handle

closed 2019-10-22 by thundron

AMDphreak, 2019-10-23:
> The "issue" here is more the fact that we don't apply any logic over the image being flashed which means whatever is in the image is read and written without any modification. If the partition table is invalid, it is such in the image itself and we decided not to mess with partitions except for specific cases (on Windows, we hold the first bytes to prevent the OS from mounting the target before it's fully flashed).
> We do appreciate the thorough analysis anyway and I guess you saw by yourself _why_ we don't modify images - because it would be a chore to do so for every image Etcher users flash (us included!), as each and every of them can be and often is different in ways that (always _for now_) we chose not to handle
> 
> Closing for now as it's currently out of scope

I don't think this is the problem. Ask yourself: Why would multiple ISO's from different vendors all have the same partition table bug? The RedHat (CentOS/Fedora) ISOs are likely burned by a different tool than the Debian-based distros' ISOs (Linux Mint). Did you inspect the ISOs available for these projects? I find it highly unlikely that they would include a random EFI partition inside another partition in the ISO. ISO 9660/Joliet doesn't have a specification for EFI either, if I recall correctly. 9600 was defined in the 70s or 80s, before EFI was a thing.

lurch, 2019-10-23:
> ISO 9660/Joliet doesn't have a specification for EFI either, if I recall correctly.

Yeah, that's why (as the link I mentioned earlier mentions) these are 'hybrid' ISOs. I don't understand the issue fully myself (so my suppositions may be incorrect), but AIUI these 'ISO' LiveLinux images support "just enough" of the ISO standard that they can be burned with standard CD/DVD writer software onto a physical CD/DVD (see [here](https://en.wikipedia.org/wiki/El_Torito_(CD-ROM_standard)) ); and they also support "just enough" of the MBR / EFI standard that they can be raw-written directly to a USB drive (e.g. by Etcher) and most BIOSes can boot from them directly.
But this 'hybrid' nature is what confuses most disk-mounting and disk-partitioning programs. AFAIK to do things "properly" each Linux distro would need to produce both a "proper" ISO (that could be burnt to a CD) and a "proper" raw IMG (that could be written to a USB stick) - but having two images would obviously lead to both additional hosting-burden and user-confusion, which I believe is why hybrid ISOs have become so popular.

But feel free to ignore my vague hand-waving and do your own more detailed technical analysis :smiley: