balena-io/etcher #2943: ei problem

adnanzooel, 2019-10-20:
$ npm run serve
npm ERR! code ENOENT
npm ERR! syscall open
npm ERR! path C:\adnan\package.json
npm ERR! errno -4058
npm ERR! enoent ENOENT: no such file or directory, open 'C:\adnan\package.json'
npm ERR! enoent This is related to npm not being able to find a file.
npm ERR! enoent

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\Adnan\AppData\Roaming\npm-cache\_logs\2019-10-20T08_27_54_605Z-debug.log

lurch, 2019-10-20:
> $ npm run serve

How is this related to Etcher?

thundron, 2019-10-21:
If you have an issue with Etcher, please fill in the issue template with as much information you can. It's hard to guess what's happening and to be honest I don't see anything related to Etcher here. Closing for now

closed 2019-10-21 by thundron